open-ephys/plugin-GUI #1: Loading/saving plugin parameters

arnefmeyer, 2016-01-28:
It would be quite useful to be able to load/save parameters for single plugins (not just the whole configuration). I couldn't find any dedicated function in the GUI. However, as I understand this can easily be done by calling the saveToXml and saveToXml and loadFromXml functions of the plugin, e.g., by adding load/save items to the context menu of a plugin (right click on plugin title bar).

@aacuevas: if this doesn't sound like an awfully bad idea I would go ahead and implement this in the GUI.

Arne

jvoigts, 2016-04-25:
we have something like this for the channel map already - would be great to add this in a general way to all plugins - see this previous issue https://github.com/open-ephys/GUI/issues/158

jsiegle, 2020-12-15:
This is related to another feature we'd like to add, which is copy/paste functionality for plugins. I added this to the Projects page here: https://github.com/open-ephys/plugin-GUI/projects/11

jsiegle, 2021-05-04:
This feature is available as of release version 0.5.4. Just right-click on the top of the editor to save/load its parameters.

closed 2021-05-04 by jsiegle