open-ephys/plugin-GUI #9: GUI crashes on fast movement of threshold slider.

kushagra06, 2016-03-07:
Hi,

I used the GUI for the first time and found an issue. In spike detector module, if the threshold slider is moved fast or is moved to 0, the GUI gets hanged. Also, the slider gets distorted and a rectangle is seen instead of the original slider (when moved to 0). It would be great if someone could help me fix this.
[image]

Thanks,
Kushagra

jsiegle, 2016-03-07:
If the spike detection threshold is at or close to zero, that electrode will generate too many spikes for the GUI to handle (since every sample will be above threshold). Can you try adding a simple check in the Slider to prevent the threshold from going below a certain value (say, 20)?

sept-en, 2016-03-07:
I've tried to reproduce the bug without any effect. It works fine for me both for GUI and plugin-GUI repositiories' builds.

Can you provide more details, please?

kushagra06, 2016-03-08:
@sept-en Construct the signal chain as in the screen shot and try to move the threshold slider to 0. The GUI would crash. (It crashes for me)

sept-en, 2016-03-08:
@kushagra06 I've tried it with the some configuration as in your screen and it was ok. 
Also I have no crashes while moving slider very fast.

aacuevas, 2016-03-08:
Could you please attach or send the xml of the saved chain and, if possible, the data you're replaying (if it's not one of our example ones)?

kushagra06, 2016-03-08:
@aacuevas I had made the signal chain that is given in the tutorial (example).

kushagra06, 2016-03-20:
I changed SpikeDetectorEditor.cpp file and now everything is fine. The GUI doesn't crash now. A simple if condition did the trick. How should I submit the patch?

jvoigts, 2016-03-20:
i'd say just make a pull request so we can have a look

kushagra06, 2016-03-21:
By mistake I kept working on the GUI version instead of the plugin-GUI and it crashes when the slider is moved to near 0. My bad! But in the plugin-GUI, when I move the threshold slider to 0, the GUI doesn't crash but still instead of the arc, a yellow rectangle is seen like the one shown in the screenshot above. Is it considered a bug?

sept-en, 2016-03-21:
@kushagra06 please, check if you are using the latest version of plugin-GUI. I tested several times and have no bug like that.
Can you maybe provide some video?

kushagra06, 2016-03-22:
@sept-en Here's a screencast. 
[Screencast 2016-03-22 05:32:39.mp4.zip](https://github.com/open-ephys/plugin-GUI/files/183289/Screencast.2016-03-22.05.32.39.mp4.zip)

anjaldoshi, 2020-02-26:
I tested the signal chain setup as specified in the screenshot on an Ubuntu machine (OP was using Ubuntu, that's why), and the threshold slider seems to work perfectly fine. No horizontal rectangle artifact showing up, and no crashes at all. As the artifact was a UI bug, it may have been fixed by 3f5f403bfe484d1a5e043bb320ca24e6c938e186 which upgraded JUCE to version 4.2.1. 

So, closing this issue because cannot reproduce the bug anymore. Please reopen if similar bug is encountered.

closed 2020-02-26 by anjaldoshi