open-ephys/plugin-GUI #34: GUI freeze on file open dialog in Ubuntu

410pfeliciano, 2016-04-26:
I'm using the open ephys plugin GUI complied from source. Commit SHA: 2b33286af546aa63e517a2cd848bfdb5f58c7e8b. I'm using ubuntu 16.04 and have produced the following issue with both the GNOME desktop environment and the Unity DE. When I try to open a configuration file, the GUI hangs and no file dialog window appears. My labmate has noticed that in the LXDE Desktop environment, the dialog will open, but will open behind the Open Ephys main window (not in focus). Maybe these two things are related.

I am therefore forced to restart the GUI after trying to load a configuration file.

malfatti, 2016-04-28:
I think I had this problem once in Gentoo and Debian, both running Openbox. Apparently this was a problem in Juce's FileChooser. Check Pierre's answer in Juce forum:

>**Pierre**
>The only solution was to replace
>#elif JUCE_MAC || JUCE_LINUX
>by
>#elif JUCE_MAC
>at line 103 of ../modules/juce_gui_basics/filebrowser/juce_FileChooser.cpp

See full discussion [here](https://forum.juce.com/t/freeze-when-opening-filechooser/16026).

This completely solves my problem for Debian, but, for some reason, not completely for Gentoo. Now on Gentoo it does not freeze, but instead of opening the dialog it just pops "No file chosen" (for Save as... option) or "No configuration selected" (for Open... option)  in the notification bar. No other dialogs seems to work, like the one to choose path to save recordings.

jvoigts, 2016-04-28:
Do we have any recipe for reliably reproducing this bug across systems? I have seen the file chooser pop up behind the main window a few times, i think mostly on newly installed systems, but it always just resolved itself after restarting the GUI so I was never able to pinpoint the root cause.

malfatti, 2016-04-28:
I don't know if this would a recipe but I tried 3 Gentoo installations on different machines and they all have this problem. I must point out that I always use the default/linux/amd64/13.0 profile and manually installed X, Openbox, gtk and qt libraries , etc. All gtk and qt applications are working fine.

410pfeliciano, 2016-04-29:
Installing GUI in Ubuntu 14.04 solved the problem. Thanks all.

closed 2016-05-02 by 410pfeliciano