open-ephys/plugin-GUI #35: Memory leak on shutdown (testing branch)

jsiegle, 2016-04-27:
On OS X, I'm getting the following message on GUI shutdown:

**Leaked objects detected: 510 instance(s) of class GlyphInfo**
**JUCE Assertion failure in juce_LeakedObjectDetector.h:95**

It happens even if I haven't added any processors—just closing the GUI immediately after running triggers it.

I poked around a little to see if I could find the source, but there wasn't anything obvious.

aacuevas, 2016-04-27:
I'm getting the same on Windows. I've been trying to hunt this one for a while. It only happens on Windows when running the debug version and through the debugger so I thought it was something more specific, but it doesn't seem so now. Any help hunting this one down will be well appreciated.

CleitonAguiar, 2016-05-18:
I got this message on GUI shutdown:

Disabling processors...
Disabling Record Node
Disabling Audio Node
Disabling Message Center
Filter Viewport enabled.

Saving window bounds.

DISABLING DATAVIEWPORT CONNECTION
*** Leaked objects detected: 510 instance(s) of class GlyphInfo
JUCE Assertion failure in juce_LeakedObjectDetector.h:95
*** Leaked objects detected: 510 instance(s) of class Path
JUCE Assertion failure in juce_LeakedObjectDetector.h:95
*** Leaked objects detected: 2 instance(s) of class CustomTypeface
JUCE Assertion failure in juce_LeakedObjectDetector.h:95
*** Leaked objects detected: 2 instance(s) of class OwnedArray
JUCE Assertion failure in juce_LeakedObjectDetector.h:95
*** Leaked objects detected: 2 instance(s) of class Typeface
JUCE Assertion failure in juce_LeakedObjectDetector.h:95

I'm using the plugin-GUI from the Master Brach (0.4.0). My OS is Ubuntu 14.04 ELS.

I'm also having an additional problem: the Intan Headstage cannot be detected. The Opal Kelly is recognized but nothing appears on the datastream slot.

sept-en, 2016-05-18:
Thanks, Cleiton!
We know about the problems with memory leaks and it's currently WIP.

CleitonAguiar, 2016-05-18:
You're welcome!

What about problems with the Head stage detection? Have you guys faced
something like this?

Thanks,

Cleiton.

CleitonAguiar, 2016-05-18:
I actually just tested the non-plugin GUI and the 32 channels Intan Head stage can be detected without any problems.

jvoigts, 2016-05-18:
That's odd - I don't recall ever seeing this problem, and the code for the rhythm api should be pretty much the same between the versions. Can you check whether you get any specific errors, or at what stage the headstage detection fails?

CleitonAguiar, 2016-05-18:
Hi Jakob,

Thanks for quickly replying.

**This what I see when drop the Rhythm FPGA to the signal chain:**

---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----

FrontPanel DLL loaded.  Built: Feb  1 2012  17:34:05

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6010LX45 with serial number 141000083Q

FPGA system clock: 100 MHz
Opal Kelly device firmware version: 3.1
Opal Kelly device serial number: 141000083QZ��
Opal Kelly device ID string: Opal Kelly XEM6010

Rhythm configuration file successfully loaded.  Rhythm version number: 1

Initializing acquisition board.
Sample rate set to 30000
Sample rate set to 30000
Number of enabled data streams: 8
Checking for connected amplifier chips...
Number of enabled data streams: 0
Sample rate set to 30000
  Adding node to graph with ID number 100

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 0
Setting buffer address to 0x21fd690
Input buffer address is 0x21fd690

**After click on the datastream slot:**

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 0

CleitonAguiar, 2016-05-18:
**This is the massage I got running the non-plugin GUI:

I think it's not detecting the Head Stage either:**

---- Intan Technologies ---- Rhythm RHD2000 Controller v1.0 ----

FrontPanel DLL loaded.  Built: Feb  1 2012  17:34:05

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6010LX45 with serial number 141000083Q

FPGA system clock: 100 MHz
Opal Kelly device firmware version: 3.1
Opal Kelly device serial number: 141000083Qf1
Opal Kelly device ID string: Opal Kelly XEM6010

Rhythm configuration file successfully loaded.  Rhythm version number: 1

Initializing acquisition board.
Sample rate set to 30000
Number of enabled data streams: 8
Checking for connected amplifier chips...
Enabled channels: 0 0 0 0 0 0 0 0 
Enabled channels: 0 0 0 0 0 0 0 0 
Enabled channels: 0 0 0 0 0 0 0 0 
Enabled channels: 0 0 0 0 0 0 0 0 
Enabled channels: 0 0 0 0 0 0 0 0 
Enabled channels: 0 0 0 0 0 0 0 0 
Enabled channels: 0 0 0 0 0 0 0 0 
Enabled channels: 0 0 0 0 0 0 0 0 
Number of enabled data streams: 0
Sample rate set to 30000
  Adding node to graph with ID number 100

Could somenone please just confirm if the headstage is not being recognized in this case too? If yes. I'll probably remove these comments from this topic and send a couple of questions through the Open Ephys google group.

Thanks.

aacuevas, 2016-05-19:
From the look of it, neither version is properly recognizing the headstage. Have you tried plugging it into a different port? You can also try to connect the headstage directly without the cable, to discard physical issues.

CleitonAguiar, 2016-05-19:
Hi Aarón,

Thank you for the simple but extremely useful tip.

I just have connected the headstage directly into the board. Indeed, now
the headstage can be recognized. So, the problem was the cable I have made.

I apologize for adding an off-topic into this Issue. Shall I remove my
comments?

Also, I have a couple of questions on making cables for this acquisition
system. How should I proceed?

jvoigts, 2016-05-19:
Sounds good - at least we know where the problem is. 
I'd say lets keep this as is and close the issue when the memory leak is fixed.

You can either post a cable related question on the forum/google group, or maybe comment on https://open-ephys.atlassian.net/wiki/display/OEW/Fine+wire+tether

in either case, it would be great if you could add a description of how to solve your problem to the wiki when you're done.

jsiegle, 2016-06-01:
The memory leak was related to font loading when using "static const Font" outside the class definition. It should be fixed now.

commit b8411b0f (2016-06-01): Fixed memory leak. Closes #35
closed 2016-06-01 by jsiegle