open-ephys/plugin-GUI #39: Error in VS for debug Gui RHA2000

carolinav24, 2016-05-06:
Hi, I´m trying to install the open-ephys in Windows 7, but when I debug the open-ephys.sln It mark 2 errors that it didnt found:
Source/Processors/DataThreads/IntanThread.cpp
Source/Processors/SourceNode/SourceNode.cpp

But I have this two sources...So, I dont know whats is going on with that.

I appreciate the help!

aacuevas, 2016-05-06:
Do you mean the RHA branch from the old monolithic repo? Which VS are you using?

carolinav24, 2016-05-06:
I download it from the RHA2200 eval branch (https://github.com/open-ephys/GUI.git).
I´m using VS 2013 ultimate version.

malfatti, 2016-05-06:
Indeed, those errors are from compiling rha2000-eval branch of non-plugin GUI.

Is there a RHA source plugin for plugin-GUI?

aacuevas, 2016-05-07:
There isn't yet. The build is effectively broken, at least for Windows. The best efficient route from here, I believe, is making it into a plugin for all architectures, I can probably have that done in a couple of days.
I'll keep you posted.

malfatti, 2016-05-09:
Thanks a lot @aacuevas! 

About the VS file being messed up, this is probably my fault :( I submitted some time ago a pull request (https://github.com/open-ephys/GUI/commit/ec85c770cea3b9a27f9086ba1c3ace438faf1102) in the rha2000-eval branch of the GUI that merged several improvements made in master. I tested and it compiles perfectly (with both RHA and FPGA working) in linux, but I didn't test other OSes. Sorry about that :( Probably replacing VS files in the current rha2000-eval by the ones in rha2000-eval at @jvoigts 's commit (https://github.com/open-ephys/GUI/commit/f3359ffbb1592f01801b8404ca08a5ea23aac919) would solve it.

Sorry again for the mess, I only started a pull request because it worked so nice in linux :)

carolinav24, 2016-05-09:
Thanks to both of you!!!

carolinav24, 2016-06-08:
Hi! any news?

jsiegle, 2020-02-25:
Closing this issue because the 16-channel RHA2000 Eval Board is not supported in the plugin-based version of the GUI.

closed 2020-02-25 by jsiegle