open-ephys/plugin-GUI #40: Data type clash between JUCE and other libraries

arnefmeyer, 2016-05-07:
It seems that there is some data type clash between juce's int64 and the int64 type used by other libraries, e.g., opencv. However, adding juce's namespace to all int64 definitions in CoreServices.h/CoreServices.cpp solves this issue, i.e. int64 -> juce::int64. Given that user-contributed plugins will likely use external libraries it seems a good idea to fix it in the next update. The functioning of the GUI is not affected by this as it is using juce's namespace by default.

aacuevas, 2016-05-08:
Ah, typedef clashes... It's one of those annoying things, specially when them resolve to exactly the same thing.
For safety and clarity we should probably explicitly add the juce namespace to all juce types on all headers used by plugins. It shouldn't be needed to do anything in most cpps, though. They already assume Juce namespace and they aren't seen in any way by the compiler when building plugins. Probably a good idea to do this before the next binary release.

FlorianFranzen, 2019-04-23:
This will not be easily to resolved, as JUCE's integer definitions also clashes with the C standard headers like ```stdint.h```/```cstdint```, at least under GCC.

closed 2022-07-30 by jsiegle