open-ephys/plugin-GUI #41: DataBuffer::addToBuffer numItems... useless, or just misunderstood?

beOn, 2016-05-11:
Unless I'm mistaken (definitely possible), the numItems argument from ```DataBuffer::addToBuffer()``` is confusing and potentially dangerous. Looking at the method's body (copied below) and calls to it throughout the code, the assumption seems to be that ```numItems``` will always be 1, as this is the case in all un-commented calls to it, and since ```buffer.copyFrom()``` gets called with ```numSamples=1```.

The problem is, if you try to call ```addToBuffer()``` with anything other than ```numItems=1``` and one sample per channel in ```data``` argument, two unfortunate things happen (I think...):

1. ```abstractFifo.prepareToWrite()``` and ```abstractFifo.finishedWrite()``` get misinformed about the number of buffer positions being used (see (2)). Unless there's something going on I'm not appreciating, this will lead to gaps in the buffer that don't get written, but do get read.

2. The for loop (see below) only reads one sample per channel into the buffer, so if the method gets called with more than that, everything beyond the first sample is quietly ignored.

I'd be happy to contribute a version of ```addToBuffer``` that can take more than one sample per channel, and to provide a separate method that unambiguously does what the current implementation does (takes one sample per channel). It seems like it'd make ```addToBuffer``` safer and easier to use, and would be reverse compatible with any code that currently uses ```addToBuffer``` correctly (which includes everything in master). Whaddya think?

```
void DataBuffer::addToBuffer(float* data, int64* timestamps, uint64* eventCodes, int numItems)
{
    // writes one sample for all channels
    int startIndex1, blockSize1, startIndex2, blockSize2;
    abstractFifo.prepareToWrite(numItems, startIndex1, blockSize1, startIndex2, blockSize2);

    for (int chan = 0; chan < numChans; chan++)
    {

        buffer.copyFrom(chan, // int destChannel
                        startIndex1, // int destStartSample
                        data + chan,  // const float* source
                        1); // int num samples
    }

    *(timestampBuffer + startIndex1) = *timestamps;
    *(eventCodeBuffer + startIndex1) = *eventCodes;

    abstractFifo.finishedWrite(numItems);
}
```

beOn, 2016-05-11:
Useless... no problematic... no, useless. So hard to decide!

aacuevas, 2016-05-11:
Behold what's probably one of the oldest classes in the entire GUI! Good catch on your part, it indeed has the issue you mention.

No need to have two different classes, though. Just adding another loop to this one and changing the input index so in case numItems>1 it actually inserts them in the buffer should be enough.

beOn, 2016-05-11:
Cool. So, I'm thinking I'll assume that the contents of ```data``` will be organized such that all samples for channel 1, then all samples for channel 2, then all samples for channel 3, etc., as opposed to sample 1 for each channel, then sample 2 for each channel, etc. Sound ok?

beOn, 2016-05-11:
To put it visually... I'm thinking:

```
[ch1_samp1][ch1_samp2] [ch2_samp1][ch2_samp2]
```
instead of:
```
[ch1_samp1][ch2_samp1] [ch1_samp2][ch2_samp2]
```

aacuevas, 2016-05-12:
Actually, for most acquisition hardware its usually the oposite, they get all the channels for sample 1, then all channels for sample 2, etc (following your second example). This is because the way they usually work is sampling all the channels (relatively) simultaneously, sending or storing them and then proceeding to the next sample. Since acquisition is one of the most time-critical subsystems, the addToBuffer method must not require any prior data transformation, so the most common approach it's preferred.
I wouldn't be against adding another method to add samples formatted in a non-muxed way, for the rare cases in which it might be better that way, though.

I'm thinking of the following: we can have both an `addToBufferMuxed` (following your second example) and an `addToBufferNonMuxed` (following your first) methods, both with the same parameters as the current one and then add a third method with prototype `void addToBuffer(float* data, int64* timestamps, uint64* eventCodes, int numItems, channelMuxType = DataBuffer::Muxed)` (where `channelMuxType` is an `enum` with both `Muxed` and `NonMuxed` options) which acts as a wrapper to call the appropriate method, while defaulting to the muxed one when no type is specified, thus being compatible with current code.

beOn, 2016-05-12:
Both the muxed and nonmuxed methods would be compatible with the current code, since both cases yield identical lists when there's only one sample per channel.

The only tweak I'd make to the wrapper call is to turn it around, make a method where the logic converges rather than diverges, since everything other than the ordering of the samples will be identical between addToBuffer implementations.

beOn, 2016-05-12:
One more little addition... it's possible that the buffer won't have enough room for the samples passed to this method. abstractFifo is smart enough to tell us this, and I won't write in values beyond what the buffer has room for, but since the return value is void there's not a great way, at the moment, to tell the caller that some of the samples weren't written.

Seems like there are two ways to approach this - either change the return to an int and return the number of samples written, or take an optional reference to an int, which'll be set to the number written on return. Any preference?

aacuevas, 2016-05-13:
Of course, the less redundant code the better. In fact I'd make the wrapper the only public method and the specific ones private.

As for indicating the number of samples actually written, I'd say a return value is best in this case.

beOn, 2016-05-25:
Check out that commit... I'll put in a PR once I've tested it.

Went in a slightly different direction. I added the chunkSize argument (and some documentation on what it does). Should address all our concerns in one implementation, without adding an enum.

Basically, in the most common case (all channels for sample 1, all channels for sample 2, etc), the chunkSize is 1.

For the extreme opposite (all samples for channel 1, all samples for channel 2, etc), the chunkSize is numItems.

If, for some reason, you read off four samples at a time per channel, or whatever other number... that's ok too.

snyderlab, 2016-05-25:
Eh, fixed something stupid. Forced the push to try and hide my shame, but github is cruel.

Tested on 986b246. Rhythm FPGA works as it did, and uses the default chunkSize=1. My own plugin, which uses numSamps>1 and chunkSize=1, also works nicely. I altered my plugging to use chunkSize=2... also works nicely. I we're ok. Submitting PR.

beOn, 2016-05-25:
Eh, snyderlab is also me, in this case.

FlorianFranzen, 2018-06-09:
It seems this issue was resolved with 502f532. Is there a reason it is still opened?

aacuevas, 2018-06-09:
No, this was indeed fixed and already tested. Thanks!

closed 2018-06-09 by aacuevas