open-ephys/plugin-GUI #42: GUI crashes when dialog in File Reader plugin is opened on some Linux systems

bburan, 2016-05-19:
To quote a post on the forum:

> the issue is probably because of a weird zenity (the native dialog box utility for many Linux variants) issue. We can build the GUI to not use zenity for choosing files, as follows:

> Navigate to the Builds/Linux folder (you might want to take a backup of the Makefiles, but don’t backup the Builds/Linux/builds/*).

> Open the Makefile and look for the lines (there should be 2) that say
`CPPFLAGS := $(DEPFLAGS) -D “LINUX=1” -D “DEBUG=1” .....`

>Now add anywhere the macro definition `JUCE_DISABLE_NATIVE_FILECHOOSERS=1`(the line may not be exactly same, you’ve to add the new macro definition to the `CPPFLAGS` variable) `CPPFLAGS := $(DEPFLAGS) -D “LINUX=1” -D “JUCE_DISABLE_NATIVE_FILECHOOSERS=1” -D “DEBUG=1”`

>    Do the same for Makefile.plugins.

 >   It’s strongly recommended that you use the Introjucer for plugin development, but it suffers from the same problem (build will pass, GUI will launch, but File Chooser will hang the GUI). You’ll have to edit the makefiles in Introjucer/Builds/Linux in exactly the same way.

 >   To make this kind-of persistent, whenever you open OE with Introjucer, just make sure the `"JUCE_DISABLE_NATIVE_FILECHOOSERS=1"` is present in the “Extra Preprocessor Definitions” of the Config tab > Linux Makefile, as shown. Now all plugins and the GUI would get this macro definition by default.

See https://groups.google.com/forum/#!topic/open-ephys/NydCGT48tF0 for the source of this quote.

malfatti, 2016-05-26:
This fixed all problems I had in gentoo (https://github.com/open-ephys/plugin-GUI/issues/34). Thanks!

arrow-, 2016-05-27:
As I've mentioned in the forum, the bug seems to have roots in the way JUCE invokes zenity. See this [thread](https://forum.juce.com/t/freeze-when-opening-filechooser/16026/15) on the JUCE forum. They've very recently managed to squash this bug and the patch is on their develop branch. I believe by the time OE switches to JUCE 4.X, it would be pulled into master _(I'll keep track of this too)_, and we can close this issue.
Till then, we have to use the non-native dialog box.

malfatti, 2016-05-27:
Curious thing - I tried previously compiling open-ephys with JUCE 4, but it didn't solve the problem. On the other hand, I tried 5 weeks ago, maybe now it is fixed :)

delcasso, 2016-11-10:
Thank you @arrow for pointing me in the good direction.

delcasso, 2016-11-10:
I got this error
<command-line>:0:1: **error:** macro names must be identifiers
BECAUSE I copied and pasted this **”** instead of **"**
-D “JUCE_DISABLE_NATIVE_FILECHOOSERS=1”
should be written as
-D "JUCE_DISABLE_NATIVE_FILECHOOSERS=1"

AGenews, 2017-12-06:
Thanks! This soloved my problems running OE on Lubuntu 16.04.3 LTS!
Best,
A.

closed 2020-02-25 by jsiegle