open-ephys/plugin-GUI #44: USB3 version of Opal Kelly FPGA not recognized by Open Ephys on Linux

bburan, 2016-05-20:
See #43 for a bugfix.

aacuevas, 2016-05-20:
To use USB3 in linux you just have to copy the libFrontPanel.so file located in /Resources/DLLs/Linux-USB3 as well as the correct bitfile from /Resources/Bitfiles
They are not copied by default, instead using an older libFrontPanel library, because newer ones link against libudev.0.so and thus will fail to load if that library is not present. Many not-so-new linux distributions, like Ubuntu 14 (one of the most popular against our userbase) do not have that library, having instead a newer version, libudev.1.so, so the GUI would not run out-of-the-box in those distributions.
The only known solution for that is to make a symlink to the newer libudev with the older name. This is quite hackish so we preferred to ship the older version of the library, which works in every Linux system, and then provide in the resources folder the newer one for anyone who wants to use USB3 on linux.

bburan, 2016-05-20:
@aacuevas Understood. In this case, it is a *documentation bug* and the instructions on the wiki need to be updated to include this information. Can you please reopen and I will submit a bugfix.

aacuevas, 2016-05-20:
I was looking at that at this moment. I know it was documented at some point, but it seems it is indeed not now. It should be in the Installation/Linux page of the wiki. I plan to re-add it later but you're right in that the bug page should be open until that.

bburan, 2016-05-20:
@aacuevas Sorry, I deleted my prior comment (the one that belongs just before yours) because I thought it may make more sense to open a new issue with a more appropriate title. For anyone that's wondering, I asked aacuevas to reopen the bug because it's now a documentation issue as my proposed fix in #43 would have caused issues for older Linux distributions.

bburan, 2016-05-20:
I have updated the wiki at https://open-ephys.atlassian.net/wiki/display/OEW/Linux. If you're happy with the edits then we can close this issue.

aacuevas, 2016-05-20:
Oh, the issue is not with older Linux distributions, is with newer ones (the ones that ship libudev1 instead of libudev0), that's the issue, FontPanel lib isn't updated for newer versions, so it works for Ubuntu 12 or older but not 14 or newer. 

I'll edit the wiki later, I cannot do it now.

bburan, 2016-05-20:
Found the place in the wiki that documents this.

https://open-ephys.atlassian.net/wiki/display/OEW/Rhythm+FPGA

Probably wouldn't hurt to include a note in the Linux install section as well though.

closed 2020-12-15 by medengineer