open-ephys/plugin-GUI #53: Write spikes from spike detector nodes

beOn, 2016-06-07:
Nodes that identify spikes should be responsible for passing them to the RecordNode, and their editors should provide the UI for controlling which spikes gets written.

Right now, spikes get passed to the record node from SpikeDisplayNode. It doesn't seem like that should be the responsibility of SpikeDisplayNode, for one because there are other spike data display nodes coming along (SpikeRaster, perhaps a TriggeredAverageNode), and it's not clear one kind of spike data displaying node in particular has the privilege of recording the data.

If it doesn't conflict with any current plans, I'd be happy to do the work on this. I think it'd roll out in two phases:

1. Move the current logic into SpikeSorter
2. Build out some UI for choosing which spikes to record. Nothing fancy at this point, just a solid foundation. I'm thinking a checkboxy-thing that's always there for 'unsorted,' then checkboxes that appear /disappear as you add/remove units. Simple but clear.

Whaddya think?

jsiegle, 2016-06-07:
This sounds like a good plan! The command for saving a spike to disk is simple (see line 287 of [SpikeDisplayNode.cpp](https://github.com/open-ephys/plugin-GUI/blob/master/Source/Plugins/BasicSpikeDisplay/SpikeDisplayNode/SpikeDisplayNode.cpp). I think this should be moved to any plugins that generate spikes (SpikeSorter and SpikeDetector). Adding some sort of graphic to GraphViewer to indicate where spikes are being saved would also be useful.

jsiegle, 2020-02-25:
We are moving all recording capabilities to a plugin-like RecordNode that will write whatever data it receives, including spikes. So instead of recording spikes via the SpikeSorter/SpikeDetector or SpikeDisplayNode, spikes will be recorded from whatever RecordNode(s) are downstream of these processors. This should simplify things by making the point in the signal chain at which spikes are being recorded more explicit.

beOn, 2020-07-23:
Good stuff. Thank you!

closed 2020-07-23 by beOn