open-ephys/plugin-GUI #60: Clicking "none" in channel map side tab silently deletes reference channel

jonnew, 2016-06-21:
OS: Ubuntu 14.04
OE Version: 9477167

In the channel map component

1. Select a reference channel using the "PARAM"  Tab in the side tab of the component
2. Go to the Audio Tab. 
3. Select a channel in Audio tab for listening through speakers.
4. Press "none" while still in Audio tab
5. Reference channel subtraction is no long applied. In the "PARAM" tab, the reference channel is still selected even though its not doing anything.

jonnew, 2016-06-21:
A suggeste

sept-en, 2016-06-21:
Hi, I've tried to do that but it's still working ok.
What plugin have you used while testing it?
(I'm testing the **development** branch now, maybe it was solved there because we made some changes to this component. However, provide some more information, if it's possible, please.
There always a chance that this bug is still presented there.)

Best regards,
Kirill Abramov.

jonnew, 2016-06-22:
I'm not sure what more information to provide. I can follow the steps I have listed above to consitantly produce the issue. Perphaps I will try the dev branch although that worries me because I'm actually doing recordings. I can provide my processing chain if that is helpful.

jonnew, 2016-06-23:
Here is the processor chain I was using. Can you see if it is fixed on your development branch when this is loaded?

[adjustment_00.zip](https://github.com/open-ephys/plugin-GUI/files/329149/adjustment_00.zip)

commit 1a832003 (2020-12-23): Fix bug in Channel Selector. Closes #60
closed 2020-12-23 by anjaldoshi