open-ephys/plugin-GUI #61: Some values are impossible to select in the PARAM Grid

jonnew, 2016-06-21:
OS: Ubuntu 14.04
OE Version: 9477167

Some channels cannot be accessed by scrolling on the side tab "PARAM" tables. For instance, look at the PARAM table for the LFP viewer in the top most position:

[image]

If I scroll, the table advances by more than two rows. This means that there are rows (e.g. 13 -17 in these images) that are not accessible for selection by clicking.

[image]

IMO, all of this would be solved by using a standard list selection element analogous to the [QT combobox](http://doc.qt.io/qt-4.8/qcombobox.html) instead of the custom tables. These elements also come with a great deal of sanity checking capabilities so you don't have to roll your own.

sept-en, 2016-06-21:
Hi!

We've already solved it in the **development** branch, the element with buttons was reimplemented and also we added possibility to hide the components for selecting the channels using slice format (textbox at the top and '+', '-' buttons).

I guess these changes will be merged into master in the new release version of Open-Ephys GUI.

Best regards,
Kirill Abramov.

jonnew, 2016-06-22:
OK, but of course, the + and - buttons, along with selecting groups of channels with the textual syntax have been _solved_ in a standard combobox. Why not use the standard solution? It even looks like JUCE has one

https://www.juce.com/doc/classComboBox#details

I would push for ditching these grids of channels in favor of comboboxes. They are much more flexible IMO and come with lots of functionality packed in that we should not have to recreate. 

Additionally, I've experienced some strangeness when selecting channles in the channel map component that may be related. When I try to drag a channel that is > 64, the mouse must hover beneath all the channel numbers (like where the channel would have been located _before_ I scrolled the grid down). Look at this image

[image]

My mouse does not show up in the screenshot, but you can see its position which is at the hovering `89`. You can see where this channel will be inserted if I was to release the left mouse button in the grid way above where the mouse is. Additionally, you can see an extra row of numbers shooting out the top of the grid.

aacuevas, 2016-06-22:
Juce's ComboBoxes are just drop-down selectors (with an optional text input), they do not allow the selection of multiple, random values (which is sometimes required in the param tab, but always in the record one) nor can they display channels in a nice grid-based way. I'm not sure if there is any Juce pre-made component that could work for the channel selectors, but it's worth checking. (Although don't keep your hopes high. With Juce its relatively easy to create UI components, but it itself only has a very scarce handful of them pre-made).

The channel mapper one is a completely separate issue, probably some bad index in the drag&drop code. I'll check it too.

jonnew, 2016-06-22:
OK, I will open a separate issue for it.

closed 2020-12-15 by medengineer