open-ephys/plugin-GUI #62: OE board's LEDs should default to off

jonnew, 2016-06-22:
When the Rhythm FPGA component is dragged to the processing chain, the LEDs on the OE board automatically turn on. The user must click the LED button to turn them off. They are very bright. If an animal is present, they become a potentially unwanted optical stimulus. The user should have to opt in to the LEDs turning on.

jsiegle, 2020-12-22:
If the LEDs are turned off in the GUI, they will remain off when loading a saved configuration. However, they do flash briefly when the plugin is loaded. @aacuevas, is this easy to fix?

aacuevas, 2020-12-22:
Those flashes are probably triggered by the first port scan the `RHD2000Thread` constructor performs, before any configuration is loaded. Disabling the LEDs prior to that call should eliminate these blinks.