open-ephys/plugin-GUI #66: Immediate Seg fault after successful compile

linussun, 2016-06-23:
Hi,
I've tried compiling the development branch of plugin-GUI and it succeeds with both make and making the plugins. However there is an immediate segmentation fault (core dumped) after the command line output shows JUCE v3.0.6 and "Created processor graph.' I'm using Ubuntu 12.04LTS (fully upgraded) and have successfully compiled and used the last non-plugin GUI 0.3.6 on my system. Any suggestions on how to debug this? What is the test system in Linux that you are using for development?
Thanks,
Linus

aacuevas, 2016-06-23:
We use mainly Ubuntu 14.04 for testing, but the GUI is known to work on Ubuntu 12 or other distros.
The crash you're getting is strange, as the stage in which the GUI is crashing for you hasn't barely changed (if at all) since 0.3.6. It seems to be crashing at the audio device manager initialization, which is weird. Have you updated any alsa or pulseaudio related packages lately? Did you build 0.3.6 with the exact same installed packages you have now or did you update after building it? If the latter, could you try rebuilding 0.3.6 to see if it still works or also crashes with an updated system?

Also, it would be very useful if you could run the gui under gdb and get a backtrace (run "gdb ./open-ephys" in the oe binary folder, then type "run" and, after the crash, type "bt" ) so we can see where exactly is it crashing.

linussun, 2016-06-27:
I did actually update some software packages on ubuntu 12.04LTS (risky I know) maybe including the audio package, I'm not sure. In any case, I tried rebuilding 0.3.6 and it does successfully compile and the binary executes just fine loading up a prior configuration as well. 

I went back to the plugin build and this is the output I got as you requested: 

linus@oep:~/git/plugin-GUI-June16/Builds/Linux/build$ gdb ./open-ephys
GNU gdb (Ubuntu/Linaro 7.4-2012.04-0ubuntu2.1) 7.4-2012.04
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
For bug reporting instructions, please see:
<http://bugs.launchpad.net/gdb-linaro/>...
Reading symbols from /home/linus/git/plugin-GUI-June16/Builds/Linux/build/open-ephys...done.
(gdb) run
Starting program: /home/linus/git/plugin-GUI-June16/Builds/Linux/build/open-ephys
warning: no loadable sections found in added symbol-file system-supplied DSO at 0x7ffff7ffa000
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
JUCE v3.0.6

Xlib:  extension "MIT-SHM" missing on display "localhost:11.0".

Created processor graph.

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff5eecb36 in vfprintf () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007ffff5eecb36 in vfprintf () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff5faa234 in __vsprintf_chk () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff5faa17d in __sprintf_chk () from /lib/x86_64-linux-gnu/libc.so.6
#3  0x00007ffff737c510 in ?? () from /usr/lib/x86_64-linux-gnu/libasound.so.2
#4  0x00007ffff737d723 in snd_device_name_hint () from /usr/lib/x86_64-linux-gnu/libasound.so.2
#5  0x00000000005f0982 in juce::(anonymous namespace)::ALSAAudioIODeviceType::enumerateAlsaPCMDevices (this=this@entry=0x109fb00)
    at ../../JuceLibraryCode/modules/juce_audio_devices/native/juce_linux_ALSA.cpp:1143
#6  0x00000000005f116d in scanForDevices (this=0x109fb00)
    at ../../JuceLibraryCode/modules/juce_audio_devices/native/juce_linux_ALSA.cpp:949
#7  juce::(anonymous namespace)::ALSAAudioIODeviceType::scanForDevices (this=0x109fb00)
    at ../../JuceLibraryCode/modules/juce_audio_devices/native/juce_linux_ALSA.cpp:933
#8  0x00000000005f20b2 in juce::AudioDeviceManager::scanDevicesIfNeeded (
    this=this@entry=0x109f780)
    at ../../JuceLibraryCode/modules/juce_audio_devices/audio_io/juce_AudioDeviceManager.cpp:343
#9  0x00000000005f40f5 in scanDevicesIfNeeded (this=0x109f780)
    at ../../JuceLibraryCode/modules/juce_audio_devices/audio_io/juce_AudioDeviceManager.cpp:336
#10 juce::AudioDeviceManager::initialise (this=0x109f780, numInputChannelsNeeded=0,
    numOutputChannelsNeeded=2, xml=0x0, selectDefaultDeviceOnFailure=<optimized out>,
    preferredDefaultDeviceName=..., preferredSetupOptions=0x0)
    at ../../JuceLibraryCode/modules/juce_audio_devices/audio_io/juce_AudioDeviceManager.cpp:188
#11 0x000000000051f678 in AudioComponent::AudioComponent (this=0x109f780)
    at ../../Source/Audio/AudioComponent.cpp:36
#12 0x00000000005bf1df in MainWindow::MainWindow (this=0x109b840)
    at ../../Source/MainWindow.cpp:47
#13 0x00000000005bf8c1 in OpenEphysApplication::initialise (this=this@entry=0x1087110,
    commandLine=...) at ../../Source/Main.cpp:88
#14 0x000000000073450d in juce::JUCEApplicationBase::initialiseApp (this=0x1087110)
    at ../../JuceLibraryCode/modules/juce_events/messages/juce_ApplicationBase.cpp:256
#15 0x000000000051af0f in juce::JUCEApplicationBase::main ()
    at ../../JuceLibraryCode/modules/juce_events/messages/juce_ApplicationBase.cpp:229
#16 0x00007ffff5ec37ed in __libc_start_main () from /lib/x86_64-linux-gnu/libc.so.6
#17 0x000000000051cd31 in _start ()
(gdb)

closed 2020-12-15 by medengineer