open-ephys/plugin-GUI #67: Error with NetworkEvents

michbad, 2016-06-24:
I'm using the latest master version of the GUI on Ubuntu 16.04.

I've created a simple chain with FileReader (sourcing one of the example files), NetworkEvents and Splitter. When I start acquisition, it works fine, I'm able to send messages using [network_events_console.py](https://github.com/open-ephys/plugin-GUI/blob/master/Resources/Python/network_events_console.py) and they show up at the bottom of the GUI. However, when I start recording, no matter if I'm connected or not, I get the following error: `terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at`
This is the Valgrind trace:
```
==4252== Process terminating with default action of signal 6 (SIGABRT)
==4252==    at 0x6AA0418: raise (raise.c:54)
==4252==    by 0x6AA2019: abort (abort.c:89)
==4252==    by 0x625984C: __gnu_cxx::__verbose_terminate_handler() (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21)
==4252==    by 0x62576B5: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21)
==4252==    by 0x6257700: std::terminate() (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21)
==4252==    by 0x6257918: __cxa_throw (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21)
==4252==    by 0x62802BE: std::__throw_out_of_range(char const*) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21)
==4252==    by 0x59FE31: at (stl_map.h:523)
==4252==    by 0x59FE31: RecordNode::process(juce::AudioSampleBuffer&, juce::MidiBuffer&) (RecordNode.cpp:459)
==4252==    by 0x6A219C: juce::GraphRenderingOps::ProcessBufferOp::perform(juce::AudioSampleBuffer&, juce::OwnedArray<juce::MidiBuffer, juce::DummyCriticalSection> const&, int) (juce_AudioProcessorGraph.cpp:226)
==4252==    by 0x6974F3: juce::AudioProcessorGraph::processBlock(juce::AudioSampleBuffer&, juce::MidiBuffer&) (juce_AudioProcessorGraph.cpp:1316)
==4252==    by 0x6AB074: juce::AudioProcessorPlayer::audioDeviceIOCallback(float const**, int, float**, int, int) (juce_AudioProcessorPlayer.cpp:129)
==4252==    by 0x60FCA7: juce::AudioDeviceManager::audioDeviceIOCallbackInt(float const**, int, float**, int, int) (juce_AudioDeviceManager.cpp:728)
```
The line in `RecordNode::process` is `int nSamples = numSamples.at(sourceNodeId);` inside a for loop going over all record channels.

This seems to be a problem with NetworkEvents, unless I am somehow creating the chain incorrectly, however it works well when only doing the acquisition.

aacuevas, 2016-06-24:
Hi,

This is because, since the chain itself does not allow processors with zero channels, the NetworkEvents creates two dummy channels but, since it is not actually filling them with data neither providing timestamps, RecordNode cannot record from them but, as a source node, their channels are set for recording by default, thus crashing.

The workaround right now is to disabel recording for the two channels of the network node (I also recommend right clicking on the merger and disabling merging data channels for it, so this issue does not propagate down the chain). I'll also add a quick patch so its channels do not get enabled for recording by default, but we'll need to think through a better way to resolve this cases, removing the need for dummy channels.

michbad, 2016-06-24:
Thank you for this explanation!

jsiegle, 2022-07-30:
The Network Events source code now lives here: https://github.com/open-ephys-plugins/network-events

If you're still experiencing problems with this plugin, please open an issue in that repository.

closed 2022-07-30 by jsiegle