open-ephys/plugin-GUI #68: open ephys process stops when trying to change save directory in Linux

jonnew, 2016-06-24:
OS: Ubuntu 14.04
OE Version: 9477167

Sometimes, when I try to open the folder selection dialog to change the directory in which data will be saved by clicking the ellipses button next to the current save path:

[image]

Open ephys stops responding. When I check the command line, I see that OE GUI has dropped into the background and no folder selection dialog  has appeared (it is not 'behind the GUI' which I think is known bug on some Linux DE's). 

```
Disabling LFP Viewer Beta
Ending animation.
Filter Viewport enabled.

[1]+  Stopped                 ./open-ephys
build[1]$ fg
./open-ephys
```
Brining the process to the foreground causes it to hang. It must be killed with `SIGQUIT`

michbad, 2016-06-24:
This is explained in #42.

jonnew, 2016-06-27:
This is probably a duplicate of #42. I will need to continue using after applying the change suggested there to confirm.

anjaldoshi, 2020-12-18:
Tested it on Ubuntu 18.04 & 20.04 and couldn't reproduce on both. Looks like it was fixed by disabling native file chooser on Linux as mentioned in #42.

closed 2020-12-18 by anjaldoshi