open-ephys/plugin-GUI #69: Segfault when AUX ADC inputs are activated after signal chain involving beta LFP viewer has been defined

jonnew, 2016-06-29:
OS: Ubuntu 14.04
OE Version: 9477167

To reproduce

1. Load the signal chain attached to the bottom of this report
2. After signal chain has loaded, enable
[ca1-recording.oesettings.zip](https://github.com/open-ephys/plugin-GUI/files/338284/ca1-recording.oesettings.zip)
 the AUX ADCs on the rhythm FPGA source node
3. Press play
4. Receive

```
...
Resizing buffer. Samples: 600000, Inputs: 139
   Enabling VisualizerEditor
Beginning animation.
Filter Viewport disabled.
[New Thread 0x7fffe9cb3700 (LWP 9746)]
[Thread 0x7fffe9cb3700 (LWP 9746) exited]
[New Thread 0x7fffe9cb3700 (LWP 9747)]
[New Thread 0x7fffe94b2700 (LWP 9748)]

Adding audio callback.

Program received signal SIGSEGV, Segmentation fault.
0x00007fffedff0b5c in LfpDisplayNodeBeta::LfpDisplayCanvas::updateScreenBuffer (this=0xb1c9e50) at LfpDisplayCanvas.cpp:1046
1046	                                samplesPerPixel[channel][sbi][c]=sample_current;
```

The following procedure fixes the segfault

1. Load processing chain
2. Remove all Beta LFP viewers
3. Enable the AUX ADCs on the rhythm FPGA source node
4. Re-add the Beta LFP viewer

[ca1-recording.oesettings.zip](https://github.com/open-ephys/plugin-GUI/files/338285/ca1-recording.oesettings.zip)

jsiegle, 2016-06-29:
I think I'm having similar issues with the Beta LFP Viewer on Windows. I'll try to track down the source of the problem later this week.

jonnew, 2016-06-29:
OK, great. Let me know if you want me to test things or provide a backtrace.

jsiegle, 2016-07-01:
There were 2 separate problems, I think I fixed both of them.

1. There was some ugly stuff going on with float array allocation for the `samplesPerPixel` variable. And then the array size wasn't being updated when the number of channels changed, hence the crash when you add the ADC channels. I changed `samplesPerPixel` from a `float***` to a `std::vector<std::array<std::array<float>>>` and things seem to be stable now. But check it and let me know.

2. `samplesPerPixel` is size channels x `MAX_N_SAMP` x `MAX_N_SAMP_PER_PIXEL`. The default was to make space for 5000 samples and 1000 samples per pixel. This becomes massive for high channel counts. I changed `MAX_N_SAMP_PER_PIXEL` to 100 and it helps, but even this seems bigger than we want it to be. @jvoigts, what do you think is the minimum size we can use and still have the histogram-based display look ok?

jonnew, 2016-07-04:
Is this pushed to the development branch?

jsiegle, 2016-07-05:
Yep, it should be there.

closed 2022-07-30 by jsiegle