open-ephys/plugin-GUI #79: Buffer size interface button not visible in release v0.4.1

jsiegle, 2016-08-26:
In the current binaries (on Mac at least), the buffer size button in the control panel is invisible (but still can be clicked). It should be an easy fix, but I wanted to make a record of it here.

aacuevas, 2016-08-26:
It doesn't show on the current Windows binaries too, but it does on my working branch, so it was probably fixed in development at some point.

sept-en, 2016-08-26:
As I remember I've changed the design of this top panel some months ago.
So now it looks like this. So, as @aacuevas mentioned it's fixed in the **development** branch.
[image]

However, I think that we should somehow accent that this is exactly a button, not just a plain text. Maybe it's better change the text of it or replace text with some configuration icon? What do you think?

Best regards,
Kirill Abramov.

mikewehr, 2016-10-11:
Can anyone point me to the bugfix that makes this button visible?

closed 2020-12-15 by medengineer