open-ephys/plugin-GUI #83: How do I build the GUI with Visual Studio 2015?

galenlynch, 2016-08-29:
I'm unable to build plugin-GUI with Visual Studio 2015 in Windows 10. I am unable to download visual studio 2013, and would like to test out some changes I've made before I submit a pull request.

I tried opening the 2013 solution and updating it, but kept getting MSBuild error MSB8020, where it could not find v140 build tools even though they were installed. I could compile a separate toy project, which suggests that the problem is with the environment variables or configuration in the project and not with my installation of Visual Studio.

Poking around at the visual studio configuration, it looks like it was generated not by visual studios but instead by introjucer/projucer? Is there some procedure for updating the visual studio project for visual studio 2015?

wehrlab, 2017-04-11:
I have similar issue with VS 2013 and other versions. If I clone the latest version of plugin-GUI, do I get updated code or do I need to do anything extra to add changes mentioned above?

galenlynch, 2017-04-29:
Hey sorry that I'm just now seeing this. My pull request was never merged, so you'll have to clone my fork if you want those changes. Note that you'll need to clone the `add-visualstudio2015-support` branch.

closed 2018-07-18 by galenlynch