open-ephys/plugin-GUI #86: Build documentation incomplete

galenlynch, 2016-08-30:
The current developer documentation [on the wiki](https://open-ephys.atlassian.net/wiki/display/OEW/Developer+documentation) doesn't describe how the build files are generated using ProJucer, which makes it hard to extend (#83). I also can't tell how the build files for the plugins are generated, which seem to also be generated by ProJucer.

aacuevas, 2016-08-30:
We had info on how to use the introjucer (now called projucer) on the old wiki, but it seems it didn't went into the new one, we'll add it again. Keep in mind, though, that changes on VS versions tend to break a lot of things, so that's why we stick to just one officially supported version. This is specially true for the plugins that require external libraries, like the KWIK format or the NetworkEvents module. You might need to manually install the appropriate libraries and modify the build files' paths accordingly.

Regarding Plugin build files, Projucer doesn't support the kind of plugins we use, so the build files were created by hand using the procedure explained in [this page](https://open-ephys.atlassian.net/wiki/display/OEW/Plugin+build+files). There is a new [plugin generator](https://open-ephys.atlassian.net/wiki/display/OEW/Plugin+Generator), based on Projucer code, which was merged to the development branch recently, but I believe the core plugins' build files haven't been reconstructed using it yet. @sept-en please correct me if I'm wrong in this.

Also, the existing build files for the plugins need a couple of tweaks due to recent changes on the juce library, so they don't compile in VS2013 neither, but that will be probably fixed in a couple of days top, until we completely recreate them.

galenlynch, 2016-08-30:
Ah ok, thanks for the explanation. My recent pull request may merit some more testing then: I've included dll's made with the new version of VS (2015) for all but the okFrontPanel.dll file, which I've tested with VS 2015 (Opal Kelly also thought they would probably work). Since I couldn't build the plugins, the other DLLs are untested.

Awesome, I didn't notice that page! Some of the environment variables in the plugin build files had identical juce visual studio variables, which is why I guess they were made with Projucer as well.

Are there plans to make the new plugin build files with VS2015? I couldn't find a way to download previous versions of VS, and due to other software we use during acquisition our lab needs to run the GUI in windows.

ethanbb, 2016-09-06:
@galenlynch, older versions of VS can be downloaded from [here](https://www.visualstudio.com/downloads/download-visual-studio-vs)!

-Ethan

galenlynch, 2016-09-09:
@ethanbb Thanks!

closed 2016-09-09 by galenlynch