open-ephys/plugin-GUI #90: How to link multiple plugins with common libraries?

ethanbb, 2016-09-06:
This issue has been brought up in PR #57, and I'm giving it a separate thread here.

In the process of developing a new plugin, I've realized that I need to give it some of the filtering functionality that is currently available to the FilterNode plugin through the [Dsp library](https://github.com/vinniefalco/DSPFilters), which lives in the FilterNode source directory. However, I don't want to duplicate code, and I'm fairly sure that copying-and-pasting this library into the source of another plugin would create namespace clashes and linking errors. Thus, I'm trying to figure out how one could share a library between multiple plugins without disrupting the plugin architecture. 

It seems that the natural solution would be to compile this library into a separate shared library file (.lib/.so/.dylib), and then have it be available when compiling the plugins that need it (and do the same with any other libraries that need to be shared). Since it would be messy to try to make that work with the way plugins are compiled now, maybe a better way is to somehow incorporate needed libraries into the main OpenEphys build? I know NetworkEvents, EventBroadcaster and I think one other plugin share ZeroMQ and get around this by requiring ZeroMQ to be installed as a prerequisite (at least on Linux?) but that obviously isn't an option for all libraries.

Thanks,
Ethan

aacuevas, 2016-09-06:
Having a plugin require a specific library code in the GUI defeats the purpose of the plugins, which is them being independent.

While this might be useful for specific code that can benefit many plugins (The Dsp library being an example. In fact I tried to have it live inside the GUI when we switched to plugins but I ran into issues doing so) it is not the solution for common code specific to a couple of individual plugins.

The ideal solution is making the common code into a shared library. It is not really messy to use them with the current plugin build system, is just a matter to add yet another dependency, as the plugins using ZMQ or HDF5 libs do, but we should find a way to streamline the creation of the build files, as that's the really dull work. @sept-en, how difficult do you think adding another build option to the plugin generator to create a simple empty library template and build project files would be?

ethanbb, 2016-09-06:
OK, I agree that adding code to the main GUI library for the benefit of specific plugins doesn't make much sense.

I didn't realize that the plugins using ZMQ and HDF5 were using shared libraries - where is that in the source code? All I can find is that in the Windows build folder, there's a ZMQ dll in the four main bin directories.

ethanbb, 2016-09-08:
To clarify, since e.g. the ZMQ plugins don't have ZMQ source code in their source directories, of course they must be using a shared library from somewhere. It looks like on Windows, when the main GUI solution is compiled, the six dlls in Resources\DLLs\VS2013(-x64) are copied into the bin directory - including the ZMQ and HD5 libraries. I assume the process is similar for compilation on other platforms. So is the plan just to prepackage any libraries that we want to be shared into a dll/so/dylib and put them in the DLLs folder?

aacuevas, 2016-09-08:
Yeah, I was refering to the dlls themselves, not the code.
For code, more than prepackage the idea is to still distribute all the code but build the shared code as a library independently of the plugin build, not unlike just building two plugins instead of one. The resulting library file would just need to be copied into the open ephys executable folder (the contents of the dll folder on windows are also copied there. On linux/mac we benefit of global libraries). Of course, if we can automate most of this so the plugin generator can automatically create library build files that also copy the built libs into their proper folder, similar as how plugin build files take care of all those nuances, it would be ideal.
For already built plugins, of course, the library files would be packaged alongside the plugins themselves.

aacuevas, 2016-10-03:
To update a bit on this, I've been working on a NWB, a new hdf5-based file format which uses the same hdf5 classes I created for KWIK, so I've gotten around wrapping those into a nice external library and creating a nice and relatively simple template for creating others. It's still a bit on the testing phase and it only works on Linux and Windows, we still need to figure out the project files and lib placement for OSX, but I think this should be the way to go for these cases.

You can find it in [the NWB branch of my personal forked repo](/aacuevas/plugin-GUI/tree/NWB) in case you want to have a look.

beOn, 2016-10-08:
Interesting... think there's a shot at fixing #89 with something along these lines?

aacuevas, 2016-10-08:
Not really. This is just a way to organize common code for different plugins, but does no change in any way the way the code links with Juce code and its abundance of templated singletons.

ethanbb, 2017-11-20:
Can I close this issue now that we have CommonLibs working? I think that more discussion about the philosophy of plugin dependencies could be useful, but my immediate problem has been solved.

ethanbb, 2018-02-06:
Also, are CommonLibs still not working on OSX @aacuevas or have you found a solution? I noticed that the HDF5 library does seem to be working as a dependency for KWIKFormat/NWBFormat in xcode, and looks like it uses the Library_Debug and Libaray_Release configuration files? But I couldn't figure out exactly how to make it work for another library. (Thanks for the help!)

aacuevas, 2018-02-08:
CommonLibs have been working on OSX for a while. What we have is proper doctumentation on how to make the build files for them, I'll need to write a page.
Short version, though, it is quite similar to making a plugin, only using the relevant Library config files instead of plugin ones. Also for OSX, instead of bundle you have to select "Library" and then "Normal (C++)" as well as skip the step in which you have to delete all configuration parameters.

ethanbb, 2018-02-08:
OK. Yeah, I inferred some of those steps, but missed the differences in how to create and configure the project in XCode. Anyway, thanks for taking care of the DSP library.

closed 2018-02-08 by ethanbb