open-ephys/plugin-GUI #91: Freeze with failure to write in OSX 10.11.6

macstry, 2016-09-20:
I recent have been trying to do longer recordings, albeit at lower acquisition rates, but keep running into the same stability problem running pluggin-GUI 0.4.1: after a variable amount of time, but usually within 2 hours, the sweep on the LFP viewer freezes though the recording time continues to increase, as if I  had hit pause. Data, however, is no longer being recorded- I lose anything after the freeze. This has happened with multiple machines (one 2012 iMac, a Mac mini) with multiple independent builds from source as well as from the binaries. This is a new problem since switching to pluggin-GUI from the older GUI. I'm not seeing any error messages, and if I hit a timestamp, it will still write to the message file- it only appears that the Ephys/EEG data is lost. Has anyone else run into this?

jsiegle, 2016-09-21:
Given that the recording clock continues to increment, it means that the GUI is still running, but the acquisition board has stopped sending data. It shouldn't have anything to do with the two plugins that failed to load.

How many channels are you trying to acquire, and is it via USB 2.0 or 3.0? There shouldn't be an issue if you're acquiring at sample rates under 30 kHz, but switching to a faster machine might help. Since it's taking a long time to freeze, it could indicate a buffer overrun on the FPGA, meaning the computer is having trouble keeping up with the rate of incoming data.

Someone else was experiencing instability in recordings, and it turned out to be due to a faulty power supply. Can you try powering the board with 5V from a benchtop power supply? Or try a beefier power supply (something like [this](http://www.digikey.com/product-detail/en/sl-power-electronics-manufacture-of-condor-ault-brands/TE30A0503B01/271-2915-ND/5400193))?

macstry, 2016-09-22:
I'm currently acquiring 140 channels at 1 kHz via USB 2.0 (older board)- these are actually epidural EEG leads, hence the lower acquisition rate. On previous versions (of GUI and OSX), I've been able to acquire 96 channels at 30 kHz for 20+ hours without problem with the same hardware, so I'm not sure that it's a hardware problem. I've ordered an alternate power supply and let everyone know if that works.

macstry, 2016-10-20:
Switched the power supply to the one suggested. Still having the same issue even with the new power supply, so that doesn't seem to be the issue.

jsiegle, 2022-07-30:
The plugin for interfacing with the acquisition board now lives here: https://github.com/open-ephys-plugins/rhythm-plugins

If you're still experiencing this problem with the latest version of the GUI, please open an issue in that repository.

closed 2022-07-30 by jsiegle