open-ephys/plugin-GUI #92: Can't compile with gcc 6

e-matteson, 2016-09-23:
I upgraded to gcc 6.2.1 and can no longer compile the GUI. I think this might be a problem upstream in JUCE. My searching turned up [this JUCE forum post](https://forum.juce.com/t/compilation-problem-on-fedora-24/18215/2) that didn't get any useful response, and [this discussion](https://aur.archlinux.org/packages/helm-git/) of what might be a similar issue affecting a different package.

Has anyone else upgraded gcc and had this issue? I'm working on downgrading to gcc 5.3, to confirm that the gcc version is what caused the problem.

The full errors are here:

```
Compiling juce_audio_formats.cpp
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/codecs/juce_FlacAudioFormat.cpp:78:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:110:
../../JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c: In function ‘long int juce::FlacNamespace::lround(double)’:
../../JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c:65:39: error: ‘long int juce::FlacNamespace::lround(double)’ conflicts with a previous declaration
 static inline long int lround(double x) {
                                       ^
In file included from /usr/include/features.h:368:0,
                 from /usr/include/sched.h:22,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_core/native/juce_BasicNativeHeaders.h:168,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:38:
/usr/include/bits/mathcalls.h:348:1: note: previous declaration ‘long int lround(double)’
 __MATHDECL (long int,lround,, (_Mdouble_ __x));
 ^
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/codecs/juce_FlacAudioFormat.cpp:78:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:110:
../../JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c: In function ‘int juce::FlacNamespace::FLAC__lpc_quantize_coefficients(const FLAC__real*, unsigned int, unsigned int, juce::FlacNamespace::FLAC__int32*, int*)’:
../../JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c:218:20: error: call of overloaded ‘lround(juce::FlacNamespace::FLAC__double&)’ is ambiguous
    q = lround(error);
                    ^
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/system/juce_StandardHeader.h:66:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/juce_core.h:143,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/juce_audio_basics.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:39:
/usr/include/c++/6.2.1/cmath:1594:3: note: candidate: constexpr long int std::lround(long double)
   lround(long double __x)
   ^~~~~~
In file included from /usr/include/features.h:368:0,
                 from /usr/include/sched.h:22,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_core/native/juce_BasicNativeHeaders.h:168,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:38:
/usr/include/bits/mathcalls.h:348:1: note: candidate: long int lround(double)
 __MATHDECL (long int,lround,, (_Mdouble_ __x));
 ^
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/system/juce_StandardHeader.h:66:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/juce_core.h:143,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/juce_audio_basics.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:39:
/usr/include/c++/6.2.1/cmath:1590:3: note: candidate: constexpr long int std::lround(float)
   lround(float __x)
   ^~~~~~
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/codecs/juce_FlacAudioFormat.cpp:78:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:110:
../../JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c:65:24: note: candidate: long int juce::FlacNamespace::lround(double)
 static inline long int lround(double x) {
                        ^~~~~~
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/codecs/juce_FlacAudioFormat.cpp:78:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:110:
../../JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c:247:20: error: call of overloaded ‘lround(juce::FlacNamespace::FLAC__double&)’ is ambiguous
    q = lround(error);
                    ^
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/system/juce_StandardHeader.h:66:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/juce_core.h:143,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/juce_audio_basics.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:39:
/usr/include/c++/6.2.1/cmath:1594:3: note: candidate: constexpr long int std::lround(long double)
   lround(long double __x)
   ^~~~~~
In file included from /usr/include/features.h:368:0,
                 from /usr/include/sched.h:22,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_core/native/juce_BasicNativeHeaders.h:168,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:38:
/usr/include/bits/mathcalls.h:348:1: note: candidate: long int lround(double)
 __MATHDECL (long int,lround,, (_Mdouble_ __x));
 ^
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/system/juce_StandardHeader.h:66:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/../juce_core/juce_core.h:143,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/../juce_audio_basics/juce_audio_basics.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.h:28,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:39:
/usr/include/c++/6.2.1/cmath:1590:3: note: candidate: constexpr long int std::lround(float)
   lround(float __x)
   ^~~~~~
In file included from ../../JuceLibraryCode/modules/juce_audio_formats/codecs/juce_FlacAudioFormat.cpp:78:0,
                 from ../../JuceLibraryCode/modules/juce_audio_formats/juce_audio_formats.cpp:110:
../../JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c:65:24: note: candidate: long int juce::FlacNamespace::lround(double)
 static inline long int lround(double x) {
                        ^~~~~~
make: *** [Makefile:627: build/intermediate/Debug/juce_audio_formats_d349f0c8.o] Error 1

```

arrow-, 2016-11-07:
Yes, this is an upstream problem with JUCE, and I get the same errors.
Instead of downgrading, I just work on a Ubuntu 14.04 VM, which is still stuck at `gcc 4.8` instead of host Manjaro (arch linux) :smile:

e-matteson, 2017-01-06:
Ok, I got it to compile by manually adding `-DHAVE_LROUND` to `CXXFLAGS` in the Makefile. This is probably a bad solution though - I don't really understand why it works or what effects it has, and I'm guessing the Makefile is automatically generated and shouldn't be edited.

Also, when I run GUI, the acquisition board isn't found, and then it segfaults when I press cancel. No idea if that's related to the compilation problem. It sounds similar to this unresolved [issue](https://github.com/open-ephys/GUI/issues/345) with the old GUI.

 What's involved in upgrading JUCE?

e-matteson, 2018-03-26:
FYI I'm now using gcc 7.3. I still have the same problem, and my solution of adding `-DHAVE_LROUND` still works.

AGenews, 2018-06-11:
I found my way to this discussion as compiling the plugin-GUI on Xubuntu 18.04 gave me the error:
>Makefile:641: recipe for target 'build/intermediate/Debug/juce_audio_formats_5a29c68a.o' failed

I found another reference to a similar problem here:
>http://forum.cabbageaudio.com/t/problems-with-compiling-cabbage-on-ubuntu-16-10/371

So I ended up following this advice:

> As a workaround you could open "./JuceLibraryCode/modules/juce_audio_formats/codecs/flac/libFLAC/lpc_flac.c" inside a text editor, search for line 60 and replace its contents with this:
static inline long int jlround(double x) {

... and it compiled fine.
Best,
Andreas

closed 2022-07-30 by jsiegle