open-ephys/plugin-GUI #93: Jack audio support?

malfatti, 2016-09-24:
Hej!

I'm using Jack connection toolkit, and I was trying to make the plugin-GUI to connect to Jack, and not to Alsa, but without success. Juce supports Jack, and even has a flag (JUCE_JACK), I just don't know where to apply it. I tried on the Makefile, but it didn't work (it compiles, but the gui still tries to connect directly to Alsa). Could you help me to understand where in the code Open-ephys is set to use Alsa, and where audio parameters like rate, buffer size and number of frames are set?

Thanks a lot!

aacuevas, 2016-09-25:
All the low-level audio device operations, like choosing the audio subsystem, are done by JUCE classes, specifically AudioDeviceManager for device setup AudioProcessorPlayer for callback generations. In out code, those are used mainly in the AudioComponent class.

The AudioDeviceManager has a built-in configuration window which can be called from the GUI by clicking on the latency text display in the control panel. The first option lets the user select the audio subsystem, so in theory the option to use Jack should be there. Isn't that the case for you?

Also, keep in mind that the audio subsystem is also the one that generates the regular callbacks for data acquisition, so there might be a performance hit by using Jack instead of ALSA.

malfatti, 2016-09-26:
Wow, I didn't notice that! Thanks! I'll just write my use case, maybe someone have a similar use.

I'm using audio devices to control almost everything in my experiments (speakers, "TTLs",  lasers, arduinos...). Thus, when the GUI connects directly to alsa (controlling the board directly), the audio board becomes unavaliable to other audio applications. 

In order to use other audio apps, the GUI should connect to an upper layer (like PulseAudio, or a dmix alsa plug... or Jack). With alsa+pulseaudio (which is the case for most linux users, I presume), when the GUI connects to alsa, it actually connects to alsa **through** pulseaudio. In my box, this lead to some troubles, like other audio apps stopped working in the middle of recordings. 

Now, with alsa+jack, and with the GUI using jack, I have a stable latency of 1.7ms, with no xruns and no instabilities in stimulation so far.

Thanks again!

closed 2016-09-26 by malfatti

crimedude22, 2016-10-11:
Hey folks, I am having a similar issue from trying to do similar things. 
How to you configure the AudioDeviceManager? I am not sure what is meant by "clicking on the latency text display in the control panel." I can't find anything that seems to match this description.
Thanks for any insight you might have!
-Mike

aacuevas, 2016-10-11:
The latency indicator is located between the "gate" slider and the "play" button. If you're using the precompiled binaries, though, there is a known bug ( #79 ) that makes it invisible but clicking on the area should bring the audio configuration window out anyway.

crimedude22, 2016-10-11:
Wow, that's an easter egg.
That button completely solved my problem. Thanks Aarón!
You might be interested to know the button is invisible on linux and mac (but not win), all compiled locally.

aacuevas, 2016-10-14:
Did you compile the master branch? The issue is fixed in the development branch. The master branch code is the same that got compiled into the binaries.