open-ephys/plugin-GUI #97: get zmq replies back into matlab?

mikewehr, 2016-10-31:
Hello, I'm using the zmqwrapper mexfile to send commands to the GUI, and I'd like a way to get the replies. For example, I'd like to know the reply to the command "GetRecordingPath." I'm using a workaround right now, where inside MyThreadFunction I fprintf the reply to a txt file, and then read the text file from within matlab. It works, but it would be much cleaner (and more robust) to return the reply through the mexfile plhs, for example. Does anybody have a suggestion for how to do this?
thanks!
-Mike

delcasso, 2016-11-01:
Hi Mike, I am trying to do something very similar. I want to use ZeroMQ to send data to matlab to do online display. We need to talk together. You can reach me at delcasso@mit.edu.

jsiegle, 2022-07-30:
Information such as the recording path can now be accessed remotely via the [Open Ephys HTTP server](https://open-ephys.github.io/gui-docs/User-Manual/Whats-new-in-version-06X.html#built-in-http-server).

If you're still having trouble getting the info that you need, please open another issue.

closed 2022-07-30 by jsiegle