open-ephys/plugin-GUI #106: GUI freezes when selecting files and directories

galenlynch, 2016-11-29:
The GUI (v 0.4.1, downloaded as a binary from your open-ephys.org site) freezes on Ubuntu 16.04 (gnome) when attempting to either select a file with the file reader source, or changing the directory using the top drawer menu.

alejoe91, 2017-01-20:
Hi,

I had the same problem and it seems a compatibility issue between JUCE and latest ubuntu releases.
I solved it following this:
[https://forum.juce.com/t/freeze-when-opening-filechooser/16026/15](url)

Basically you just need to replace 
#elif JUCE_MAC || JUCE_LINUX
by 
#elif JUCE_MAC
at line 103 of ../modules/juce_gui_basics/filebrowser/juce_FileChooser.cpp

I know it is not a final fix...but at least it works.

galenlynch, 2017-01-20:
Thanks for the tip!

closed 2018-07-18 by galenlynch