open-ephys/plugin-GUI #107: Documentation of XML files?

galenlynch, 2016-12-16:
I'm trying to understand the settings.xml and Continuous_Data.openephys xml files produced by the plugin gui. There are a number of XML elements and parameters that are not in the [data format specification](https://open-ephys.atlassian.net/wiki/display/OEW/Open+Ephys+format) nor in the [interfaces and standards](https://open-ephys.atlassian.net/wiki/pages/viewpage.action?pageId=3211297) wiki page. I tried look in the source code of OriginalRecording.cpp and .h, but couldn't find any other helpful information there.

Specifically, I am wondering about the following parameters:

in the Continuous_Data.openephys file:

- `number` parameter of `EXPERIMENT` element
- `separatefiles` parameter of `EXPERIMENT` element
- `number` parameter of `RECORDING` element (is this the same as the recording numbers in the binary files?)

in the settings.xml file file:

- `SampleRate` parameter in the `EDITOR` child of "Sources/Rhythm FPGA" `PROCESSOR` element.

Is there always only one experiment? Is a new `RECORDING` element made each time the record button is pressed in the GUI, or does the `RECORDING` element span multiple recordings?

jsiegle, 2016-12-20:
The rules for file creation are as follows:
- When recording is stopped, but acquisition continues, the GUI increments the `RECORDING` number. For Open Ephys data format, it appends the data within the same file; for the binary format, it creates a new file; for the KWIK format, it creates a new group within the same file
- When acquisition is stopped, the GUI increments the `EXPERIMENT` number. This always results in the creation of a new file the next time recording begins.
- If the "new directory" button is pressed (or the GUI is closed and re-opened), `RECORDING` and `EXPERIMENT` variables will be reset, and all files will be placed in a newly created directory.

@aacuevas, do you have anything to add?

As far as the `SampleRate` goes, the acquisition board sample rate can be updated by the Rhythm FPGA editor, so that's where that parameter is stored.

galenlynch, 2016-12-20:
@jsiegle Thanks for the explanation! It's a great help.

aacuevas, 2016-12-21:
Just a couple of extra notes:
In the Open Ephys data format, different recordings can be stored in different files if the "separate files" option is set (you can find record engine options, for those who have any, by clicking the "R" button next to the record format selection dropbox)

Regarding the Continuous_Data.openephys file itself, it's a never quite finished project to have a easily readable index of where the data is in the different files when recording with the Open Ephys file format, as well as which files pertained to each recording, in case an application needed to know how many recordings where in each file without having to read all the files in a directory. It was never fully tested, so its info might not be accurate or even relevant, and it is only created when recording in the Open Ephys format.

galenlynch, 2016-12-22:
Thanks for the explanation guys, I have tried to put this information to use in my [Julia package](https://github.com/galenlynch/OpenEphysLoader.jl) for reading Open Ephys files, and I'll try to add it to the Open Ephys wiki sometime in the near future. In terms of versioning, I noticed that `settings.xml` has at least one difference between the plugin and non-plugin GUIs (the plugin_api_version element under info). Is it safe to assume that the format of these metadata files will be factored in to Open Ephys GUI version number?

galenlynch, 2017-01-04:
Just to separate my question from the rest of my post: will these xml files be factored into GUI version numbers?

jsiegle, 2017-01-13:
Yes, we won't make any changes to the XML files without updating the version number, at least for the master branch/compiled binaries. There may be changes that occur in the development branch, but those won't be merged into the master branch without a version number increment.

galenlynch, 2017-01-13:
Thanks for the clarification.

closed 2017-01-13 by galenlynch