open-ephys/plugin-GUI #109: Can't upload bitfile to XEM6010

Jimbles, 2017-01-24:
Hi,
I am trying to use an acquisition board v2.1 (rev3) with an Opal Kelly XEM6010-LX150 on windows 10 64bit. I followed the installation instructions from the [wiki](https://open-ephys.atlassian.net/wiki/display/OEW/Windows) and the Front Panel Device seems to be recognised OK as it shows up in device manager.

However, I am getting an error saying **FPGA bitfile not found** saying the rhd2000.bit was not found in the directory. The output of the console (below) suggests that maybe it *was* found, but it didn't receive confirmation of the upload?
 
```
Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6010LX150 with serial number 1243000461

FPGA system clock: 100 MHz
Opal Kelly device firmware version: 3.1
Opal Kelly device serial number: 1243000461
Opal Kelly device ID string: Opal Kelly XEM6010

FPGA configuration failed: FPGA DONE signal did not assert after configuration.
Couldn't upload bitfile from C:\Users\Jimbles\Downloads\windows-64-master\rhd2000.bit
```

We have tried this on 2 PCs (both windows 10 64bit though) and have had no luck. The only other thing that is different is the 5V power supply, which is a 2250mA power supply from Maplin, *not* the one supplied with the board.

aacuevas, 2017-01-24:
Hi,

The issue is that the bitfile is compatible with the XEM6010-LX45, not with the LX150. 

FPGA configuration files (bitfiles) are always created for one specific device each. There's no work around that since that's just how FPGA technology works. The XEM6010-LX45 features a Xilinx XC6SLX45-2C FPGA, while the LX150 features a XC6SLX150-2C which are different devices and, as such, need a bitfile specifically created for each one.

It should be possible to just take the existing Rhythm firmware code and build it for a LX150 with little changes, as only the target device and, possibly, the devices I/O pin designations would need to be different. However, only the LX45 device is supported by the free version of Xilinx development tools, the LX150 (a larger device, with more available logic blocks) requires a paid license, of which I think we don't have one ourselves.

Jimbles, 2017-01-24:
Hey! Thank you for getting back to me so quickly!
Oh I hadn't appreciated the different versions of the XEM6010, and I totally get why it wasnt working now! D'oh!

This acquisition board (and FPGA module) was lent to us by another group, who as far as I'm aware were using it fine in experiments a year or two ago. I guess they must have compiled it themselves? Or could it be possible that an older version (c.2013) was once compiled for the LX150?

Jimbles, 2017-01-26:
That was in fact the case, and we have their compiled version for the LX150 and it all seems to be working! Is there anything that has changed in the official LX45 firmware that might cause us problems with the newer GUIs or plugins?

aacuevas, 2017-01-26:
The firmware is not something we upgrade often, in fact last update was almost a year ago, and it was a minor one. However since 2013 we've upgraded the firmware to match Intan's 1.41 and added a couple of features. Things that will not work or might not work properly include:
-Toggling the LEDs on the board (also the led patterns will be different)
-Some features regarding board DAC control (like the HPF)
-TTL hardware fast settle toggling
-Support for 64 channel headstages
-Sampling sync output signal through the BNC

Basic acquisition with 32 channel headstages should work, though, that basic functionality hasn't changed,

Jimbles, 2017-01-26:
Great! Thanks very much. We are trying to just evaluate the system to see if it meets our needs, so basic 32 channel stuff is fine for now.

closed 2017-01-26 by Jimbles