open-ephys/plugin-GUI #110: Acquisition board not found (Ubuntu 16.04, XEM6010-LX45)

tsgouvea, 2017-01-25:
After unzipping a precompiled Linux binary on my Ubuntu 16.04, copying `40-open-ephys.rules` to its place and restarting `udev`, I opened GUI and tried adding the Rhythm FPGA module to the signal chain. The FPGA (XEM6010-LX45) is connected to a USB2.0 port, but it's not being detected. Error messages are:
On GUI:
```
ACQUISITION BOARD NOT FOUND.
An acquisition board could not be found. Please connect one now.
[OK] [Cancel]
```
On console:
```
FrontPanel DLL loaded.  Built: Feb  1 2012  17:34:05

Scanning USB for Opal Kelly devices...

Found 0 Opal Kelly devices connected:

Device could not be opened.  Is one connected?
```
Yes, it is. But indeed, the FPGA doesn't show on the `lsusb` list (cable and USB port work fine.)  Strangely, when I connect it to a USB3.0 port, it is detected, but I get a different error:
```

FrontPanel DLL loaded.  Built: Feb  1 2012  17:34:05

Scanning USB for Opal Kelly devices...

Found 1 Opal Kelly device connected:
  Device #1: Opal Kelly XEM6010LX45 with serial number 00000004YU

FPGA system clock: 100 MHz
Opal Kelly device firmware version: 3.1
Opal Kelly device serial number: 00000004YUï¿½
Opal Kelly device ID string: Opal Kelly XEM6010

FPGA configuration failed: FPGA DONE signal did not assert after configuration.
Couldn't upload bitfile from /home/thiago/Programs/open-ephys-linux/rhd2000.bit
```
A dialog window opens, and pointing to either `rhd2000.bit` or `rhd2000_usb3.bit` leads to the same error. Anything I'm doing wrong?
Help appreciated.

aacuevas, 2017-01-26:
If the board is a XEM6010 it should work on both USB2 and USB3 ports, both with the rhd2000.bit file (since the board is still the usb2 one). I find it quite weird that it works on a usb3 port but not on a usb2 one. Also, the serial number it shows doesn't seem right. I worry it might even be a fault in the hardware.

I'll check if there is some particular issue with ubuntu16 that we haven't noticed, but please try the following in the meanwhile:
-Is the power wall adapter the one included by us? If not, please check the power rating and, if you have a multimeter, check that it actually outputs 5V
-Try plugging the wall adapter in both power plugs present in the board, in case one was faulty.
-Try on a different computer, if you have one available, to see if the same issue is present on it.
-Try plugging the board to different ports and check if the issue is reproducible among the different USB2 and 3 ports
-Please run lsusb with verbose option when the board is connected on ports that recognize it and send me the lines relevant to the FPGA (if for some reason different ports yielded different results on the id fields, please send them to me too).

tsgouvea, 2017-01-26:
> -Is the power wall adapter the one included by us? If not, please check the power rating and, if you have a multimeter, check that it actually outputs 5V

I'm using a generic power adapter (nominal output 5.0V/1.0A, measured 5.09V/0.98A). Using one I believe was shipped with the board (measured 5.19V/0.98A) changed nothing.

> -Try plugging the wall adapter in both power plugs present in the board, in case one was faulty.

Tried, no change.

> -Try plugging the board to different ports and check if the issue is reproducible among the different USB2 and 3 ports

Reproducible for different ports of the same machine, but not for different machines.

> -Try on a different computer, if you have one available, to see if the same issue is present on it.

It is actually listed on `lsusb` on a different Ubuntu 16.04 machine, whether plugged on USB2 or USB3 port. Could it be a driver issue (eg. present/absent, different version)?

> -Please run lsusb with verbose option when the board is connected on ports that recognize it and send me the lines relevant to the FPGA (if for some reason different ports yielded different results on the id fields, please send them to me too).

Below, the output of `lsusb -v` from the machine where I need it to work, when plugged on the usb3 port.

```
Bus 003 Device 006: ID 151f:002d  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol       255 
  bMaxPacketSize0        64
  idVendor           0x151f 
  idProduct          0x002d 
  bcdDevice            3.01
  iManufacturer           1 Opal Kelly
  iProduct                2 Opal Kelly XEM6010-LX45
  iSerial                 3 00000004YU
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol       255 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)
```

In addition, I tried a different Open Ephys board/FPGA set that I was told is functioning well. Two issues present only on the original board hint at a possible hardware issue: (1) the fan is always on at high speed as soon as I plug the power cable, and (2) the -5V pin right next to GND_6 (bottom right of acquisition board) measures to actual +3.26V. However, results regarding USB detection were identical (except for Device and iSerial numbers), meaning the issue persists even after I replaced the entire open-ephys system.

cjblack, 2017-11-08:
Running into the same exact issue with Ubuntu 16.04. The acquisition board I am using seems to work fine on a windows computer. I am thinking it has something to do with how Ubuntu was installed...still trying to work through this. Any updates?

jsiegle, 2022-07-30:
It seems like the Rhythm FPGA plugin is generally working fine on Linux now. If you're still experiencing this problem with the latest version of the GUI, please open an issue in the new [Rhythm Plugins](https://github.com/open-ephys-plugins/rhythm-plugins) repository.

closed 2022-07-30 by jsiegle