open-ephys/plugin-GUI #112: [Development branch] OpenEphysHDF5Lib.so errors

malfatti, 2017-01-26:
Hello!

After compiling the GUI (with no errors), when I load the GUI I got the following errors:
```bash
[...]
Loading Plugin: NWBFormat... ../../Source/Processors/PluginManager/PluginManager.cpp:177: Failed to load plugin DLL: OpenEphysHDF5Lib.so: cannot open shared object file: No such file or directory
 DLL Load FAILED
[...]
Loading Plugin: KWIKFormat... ../../Source/Processors/PluginManager/PluginManager.cpp:177: Failed to load plugin DLL: OpenEphysHDF5Lib.so: cannot open shared object file: No such file or directory
 DLL Load FAILED
[...]
```
As a consequence, I can only save data in OpenEphys and Binary formats. 
In the build folder there is an OpenEphysHDF5Lib.so file:
```bash
$ ls build/
intermediate    libokFrontPanel.so   open-ephys.so  rhd2000.bit
lastConfig.xml  OpenEphysHDF5Lib.so  plugins        windowState.xml
```
Should I link this file somewhere before/after compiling?

HDF5 libraries are working fine in my system. I can read and write hdf5 files, and stable GUI compiles and saves data in KWIK format. Here is some info on the system:
```bash
$ sudo find / -type f -iname libhdf5*
/usr/lib64/libhdf5hl_fortran.so.10.0.3
/usr/lib64/libhdf5_cpp.so.13.0.0
/usr/lib64/libhdf5.so.10.2.1
/usr/lib64/libhdf5_hl_cpp.so.11.1.0
/usr/lib64/libhdf5_fortran.so.10.0.4
/usr/lib64/libhdf5.settings
/usr/lib64/libhdf5_hl.so.10.1.1

$ uname -a
Linux LeaoLabGentoo 4.8.14-rt-rt9 #2 SMP PREEMPT RT Tue Jan 17 12:33:17 BRST 2017 x86_64 Intel(R) Xeon(R) CPU E3-1280 V2 @ 3.60GHz GenuineIntel GNU/Linux
```

Thanks!

aacuevas, 2017-02-08:
Hi,

Try first to remove the OpenEphysHDF5Lib.so file and rebuilding it using `make -f Makefile.plugins libs`.

if it doesnt fix the issue, please try running the gui with the command `LDLIBRARY_PATH=. ./open-ephys` (I noticed that your file is named open-ephys.so, the .so souldn't be there, we must have itnroduced a bug in the makefile at some point, just rename it. It isn't important, though)

malfatti, 2017-02-12:
Hi @aacuevas,

Thanks for your answer! Removing and rebuilding the library didn't help, but specifying the path before the command solved the issue =)

Cheers

aacuevas, 2017-02-12:
I'll have to check the build options then, as it should look into the executable path by default.

AGenews, 2018-06-11:
... having the same issue with the current master branch. **LD_LIBRARY_PATH=/path/to/OpenEphysHDF5Lib.so ./open-ephys** helps though.
Is there a permanet fix? Changing the Makefile.plugins?
Thx & Best,
Andreas

aacuevas, 2018-06-11:
This should not happen anymore. The one responsible for looking for libraries is the main executable, and we added a linker flag to the makefile to ensure the current directory was looked into.
Which OS are you using? Can you execute `objdump -x open-ephys | grep RPATH` and paste the resulting line?

AGenews, 2018-06-12:
Dear Aarón,
I'm using Xubuntu 18.04 and unfortunately `objdump -x open-ephys | grep RPATH` returns nothing.
Cheers,
Andreas

aacuevas, 2018-06-20:
Um, weird. I'll have to check that particular distribution to see if either gcc or the linker behave differently there. Please execute just `objdump -x open-ephys` and send me its output to aaron(at)open-ephys.org

Just to be sure, if you check the Makefile there is a `-rpath,'$$ORIGIN'` flag inside the LDFLAGS  line, right?
Please also try adding that flag to the Makefile.plugins in case it did something, but according to what I know about the linker, only the executable should care about library paths.

closed 2020-12-15 by medengineer