open-ephys/plugin-GUI #113: .cpp files could use brief descriptions

mspacek, 2017-02-06:
I'm trying to figure out what files do what in the source, and I've noticed that most or all of the `.cpp` files have the same generic GPL message at the top. Maybe this is my Python side talking, but it would be really useful to have a brief description of what each file does/is for in a one- or two-line comment at the top, especially for newcomers.

mspacek, 2017-02-06:
Hm, it seems that some (but not all) of the `.h` files have some class and method docstrings. It would still be nice to have a one-liner near the top of each `.cpp` file. Not sure if that would go against some C++ convention...

aacuevas, 2017-02-08:
The convention is to put the class and method documentation on the header files, as are the ones that are referenced by external modules, while the cpp files comments are limited to implementation details, since each implementation (cpp) has a definition (h) associated (the reverse is not always true).

That said, we lack in a big way in the documentation department, is something we ought to solve and standardize some day.

jsiegle, 2022-07-30:
In the latest version of the GUI, almost every class and method includes a docstring in the header file.

closed 2022-07-30 by jsiegle