open-ephys/plugin-GUI #114: [Development branch] ExampleProcessor is incomplete

mspacek, 2017-02-06:
The wiki[ development tutorial](https://open-ephys.atlassian.net/wiki/display/OEW/Tutorial%3A+Add+a+custom+processor) refers to `ExampleProcessor` as a template to use for new processors/plugins (not clear to me what the difference is). This seems to be outdated, probably referring to the old [GUI ](https://github.com/open-ephys/GUI). Currently, the development branch of the plugin-GUI has a `Source/Plugins/ExampleProcessor` folder that is empty, save for `ExampleEditor.cpp`, which itself refers to `ExampleEditor.h` and `ExampleProcessor.h`, which are missing from the project. Presumably, `ExampleProcessor.cpp` should be there as well. I take it copying those files from the original [GUI ](https://github.com/open-ephys/GUI) wouldn't work?

This seems like a critical piece of documentation to get right, for newbies to quickly get started writing their own plugins.

Aha, I just noticed that in master, all the files are there. Did something go wrong in the development branch?

aacuevas, 2017-02-08:
It seems they got moved to the template folder when we added the plugin generator functionality, but it would be good to still have them in their reference folder for people who want to make their plugins manually themselves.

Thanks for the catch!

FlorianFranzen, 2018-06-09:
This issue by now is also present in master branch.

jsiegle, 2021-05-04:
We have removed the ExampleProcessor from the main repository. We now recommend that people base new plugins off the template available here: https://github.com/open-ephys-plugins/oeplugin

closed 2021-05-04 by jsiegle