open-ephys/plugin-GUI #118: Segfault on UP/DOWN keypress in unhighlighted processor

mspacek, 2017-02-17:
I've noticed that sometimes hitting a navigation key (up or down) on the keyboard when one of the processors has focus, can cause a segfault. The contents of the signal chain doesn't seem to matter. To reproduce:

1. drag any block onto the signal chain, even just a splitter
2. click inside of it, so it loses its yellow selection highlight around the border
3. hit UP or DOWN
4. segfault

This is in Xubuntu 16.04.

Here's the printout when I run it in gdb. I'm not familiar with that section of code:
```
Thread 1 "open-ephys.so" received signal SIGSEGV, Segmentation fault.
0x00000000005f1f84 in EditorViewport::keyPressed (this=0x1240f90, key=...) at ../../Source/UI/EditorViewport.cpp:642
642	            lastEditorClicked->switchIO(1);
(gdb) l
637	
638	            return true;
639	        }
640	        else if (key.getKeyCode() == key.downKey)
641	        {
642	            lastEditorClicked->switchIO(1);
643	            return true;
644	        }
645	    }
646	
(gdb) 
```

mspacek, 2017-02-17:
Strange, it looks like switchIO() is only meant for Splitters and Mergers, yet it's still called for other kinds of Processors, like CAR. I get the same segfault if I use a CAR Processor.

aacuevas, 2017-02-17:
switchIO() is meant only for splitters and mergers, but it is defined for other kind of processors as an empty method.

The crash is caused by lastEditorClicked being null when no editor is highlighted. I'll add a simple check to avoid trying to call a method from a null pointer. 

Thanks for the catch!

mspacek, 2018-05-11:
Hello! Was this ever fixed? I just built the latest dev branch, and it still crashes when I follow the above instructions. Here's the gdb output for a CAR processor:
```
[New Thread 0x7fffdf4f1700 (LWP 22887)]
Item dropped at insertion point 0
Creating from description...
Filters::Common Avg Ref (1-6)
  Adding node to graph with ID number 100

Common Avg Ref updating settings.
Common Avg Ref setting num outputs to 0

Thread 1 "open-ephys" received signal SIGSEGV, Segmentation fault.
0x00000000005ea144 in EditorViewport::keyPressed (this=0x11fd930, key=...) at ../../Source/UI/EditorViewport.cpp:649
649	            lastEditorClicked->switchIO(1);
(gdb) l
644	
645	            return true;
646	        }
647	        else if (key.getKeyCode() == key.downKey)
648	        {
649	            lastEditorClicked->switchIO(1);
650	            return true;
651	        }
652	    }
653	
(gdb) 
```

commit 4da3bc25 (2020-12-23): Only splitters + mergers respond to up/down keys. Closes #118.
closed 2020-12-23 by anjaldoshi