open-ephys/plugin-GUI #122: Segfault in BinaryRecordingEngine::SequentialBlockFile when .dat file reaches 128.0 GiB

mspacek, 2017-03-01:
As mentioned [here](https://groups.google.com/forum/#!msg/open-ephys/8mbdNqOo01Y/1kl4a-ItDQAJ), after doing a 32 channel test recording for about 20 hours to a .dat file, I get a segfault. This time, I caught it in gdb:
```
Thread 13 "Record Thread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffdfffe700 (LWP 11693)]
BinaryRecordingEngine::SequentialBlockFile::writeChannel (this=0x7fffb8000be0, startPos=<optimized out>, channel=0, 
    data=0x5bbcf80, nSamples=1024) at SequentialBlockFile.cpp:84
84			int16* blockPtr = m_memBlocks[bIndex]->getData();
(gdb) l
79		int startIdx = startPos - m_memBlocks[bIndex]->getOffset();
80		int startMemPos = startIdx*m_nChannels;
81		int dataIdx = 0;
82		while (writtenSamples < nSamples)
83		{
84			int16* blockPtr = m_memBlocks[bIndex]->getData();
85			int samplesToWrite = jmin((nSamples - writtenSamples), (m_samplesPerBlock - startIdx));
86			for (int i = 0; i < samplesToWrite; i++)
87			{
88				*(blockPtr + startMemPos + channel + i*m_nChannels) = *(data + dataIdx);
(gdb) 
```
Have I perhaps hit some signed 32-bit limit? Note that `2**31*2*32/1024/1024/1024 == 128.0`. The file size of the .dat is just shy of that, 137,438,691,328 bytes == 127.999755859375 GiB, probably because it didn't get around to writing the block that pushed it over the limit. I have made substantial changes to the BinaryWriter, (see https://github.com/open-ephys/plugin-GUI/compare/development...mspacek:blab) but none of those have been to the actual code in SequentialBlockFile.cpp that writes to the .dat file.

I still have it sitting in gdb, so I can run some more debugging commands if that would help. Replicating this is obviously very time consuming...

aacuevas, 2017-03-01:
Hi,

Can you get the full backtrace as well as the value if the bIndex variable?

mspacek, 2017-03-02:
Sure, here is the full backtrace:
```
(gdb) backtrace full
#0  BinaryRecordingEngine::SequentialBlockFile::writeChannel (this=0x7fffb8000be0, startPos=<optimized out>, 
    channel=0, data=0x5bbcf80, nSamples=1024) at SequentialBlockFile.cpp:84
        bIndex = 1
        writtenSamples = 0
        startIdx = <optimized out>
        startMemPos = <optimized out>
        dataIdx = <optimized out>
#1  0x00000000005b8c3b in RecordThread::writeData (this=this@entry=0x11fe570, dataBuffer=..., 
    maxSamples=maxSamples@entry=4096, maxEvents=maxEvents@entry=32, maxSpikes=maxSpikes@entry=32, 
    lastBlock=lastBlock@entry=false) at ../../Source/Processors/RecordNode/RecordThread.cpp:124
        eng = 0
        chan = 0
        timestamps = {data = {<juce::DummyCriticalSection> = {<No data fields>}, elements = {data = 0x7fffb8043090}, 
            numAllocated = 32}, numUsed = 32}
        idx = {data = {<juce::DummyCriticalSection> = {<No data fields>}, elements = {data = 0x7fffb8041c50}, 
            numAllocated = 32}, numUsed = 32}
        events = std::vector of length 0, capacity 0
        spikes = std::vector of length 0, capacity 0
#2  0x00000000005baa20 in RecordThread::run (this=0x11fe570) at ../../Source/Processors/RecordNode/RecordThread.cpp:97
        dataBuffer = <optimized out>
        closeEarly = false
#3  0x0000000000773d77 in juce::Thread::threadEntryPoint (this=0x11fe570)
    at ../../JuceLibraryCode/modules/juce_core/threads/juce_Thread.cpp:102
        currentThreadHolder = {referencedObject = 0x7fffd80008c0}
#4  0x0000000000773f69 in juce::juce_threadEntryPoint (userData=<optimized out>)
    at ../../JuceLibraryCode/modules/juce_core/threads/juce_Thread.cpp:117
No locals.
#5  juce::threadEntryProc (userData=<optimized out>)
    at ../../JuceLibraryCode/modules/juce_core/native/juce_posix_SharedCode.h:903
No locals.
#6  0x00007ffff6c536ba in start_thread (arg=0x7fffdfffe700) at pthread_create.c:333
        __res = <optimized out>
        pd = 0x7fffdfffe700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140736951478016, 4371542949540180860, 0, 140737488343967, 
                140736951478720, 0, -4371472581841865860, -4371523349137935492}, mask_was_saved = 0}}, priv = {pad = {
              0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#7  0x00007ffff5ee082d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
No locals.
(gdb) 
```
From above, it looks like `bIndex == 1`. I guess I was in the wrong frame when I tried to do this
```
(gdb) print bIndex
No symbol "bIndex" in current context.
```

mspacek, 2017-03-03:
I re-ran it overnight, and this time printing bIndex worked:
```
(gdb) print bIndex
$1 = 1
(gdb) 
```

mspacek, 2017-03-13:
Here are a couple more variables printed out:
```
(gdb) print writtenSamples
$9 = 512
(gdb) print nSamples
$10 = 1024
(gdb) 
```
Too bad `startPos` only shows up as `<optimized out>`. Is there some compile flag I can set to prevent this? I'm not sure if debugging is enabled by default in the makefile. [Apparently](http://stackoverflow.com/a/1354770) one can give g++ the `-O0' argument to disable optimizations.

mspacek, 2017-03-13:
Unfortunately, when I try to change `-O3` to `-O0` in both `Makefile` and `Makefile.plugins`, the linker gives an undefined references error:
```
$ make
...
Linking open-ephys
build/intermediate/Debug/Parameter_b3e5ac9e.o: In function `generateCodeForParameter(Parameter const&)':
/home/mspacek/src/open-ephys/Builds/Linux/../../Source/Processors/Parameter/../../../PluginGenerator/Source/Utility/openEphys_pluginHelpers.h:280: undefined reference to `CodeHelpers::indent(juce::String const&, int, bool)'
/home/mspacek/src/open-ephys/Builds/Linux/../../Source/Processors/Parameter/../../../PluginGenerator/Source/Utility/openEphys_pluginHelpers.h:292: undefined reference to `CodeHelpers::indent(juce::String const&, int, bool)'
/home/mspacek/src/open-ephys/Builds/Linux/../../Source/Processors/Parameter/../../../PluginGenerator/Source/Utility/openEphys_pluginHelpers.h:312: undefined reference to `CodeHelpers::indent(juce::String const&, int, bool)'
/home/mspacek/src/open-ephys/Builds/Linux/../../Source/Processors/Parameter/../../../PluginGenerator/Source/Utility/openEphys_pluginHelpers.h:326: undefined reference to `CodeHelpers::indent(juce::String const&, int, bool)'
/home/mspacek/src/open-ephys/Builds/Linux/../../Source/Processors/Parameter/../../../PluginGenerator/Source/Utility/openEphys_pluginHelpers.h:333: undefined reference to `CodeHelpers::indent(juce::String const&, int, bool)'
build/intermediate/Debug/Parameter_b3e5ac9e.o:/home/mspacek/src/open-ephys/Builds/Linux/../../Source/Processors/Parameter/../../../PluginGenerator/Source/Utility/openEphys_pluginHelpers.h:352: more undefined references to `CodeHelpers::indent(juce::String const&, int, bool)' follow
collect2: error: ld returned 1 exit status
Makefile:160: recipe for target 'build/open-ephys.so' failed
make: *** [build/open-ephys.so] Error 1
```

mspacek, 2017-03-14:
Progress! Even though I couldn't compile the main project with `-O0`, I *was* successful in compiling all of the plugins (of which BinaryWriter is one) with `-O0`. After running it overnight, it hit the 128.0 GB file size barrier, segfaulted, and now gdb shows something much more informative. Seems the problem is one level down, in `juce::HeapBlock::getData() `
```
Thread 13 "Record Thread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffdfffe700 (LWP 17836)]
0x00007fffefa36794 in juce::HeapBlock<short, false>::getData (this=0x0)
    at /home/mspacek/src/open-ephys/Builds/Linux/../../JuceLibraryCode/modules/juce_core/memory/juce_HeapBlock.h:163
163	    inline ElementType* getData() const noexcept                            { return data; }

(gdb) backtrace full
#0  0x00007fffefa36794 in juce::HeapBlock<short, false>::getData (this=0x0)
    at /home/mspacek/src/open-ephys/Builds/Linux/../../JuceLibraryCode/modules/juce_core/memory/juce_HeapBlock.h:163
No locals.
#1  0x00007fffefa3852a in BinaryRecordingEngine::FileMemoryBlock<short>::getData (this=0x0) at FileMemoryBlock.h:47
No locals.
#2  0x00007fffefa37b10 in BinaryRecordingEngine::SequentialBlockFile::writeChannel (this=0x7fffb8000be0, 
    startPos=18446744071562067456, channel=0, data=0x5bbefb0, nSamples=1024) at SequentialBlockFile.cpp:84
        blockPtr = 0x7fffb80013d0
        samplesToWrite = 512
        bIndex = 1
        writtenSamples = 512
        startIdx = 0
        startMemPos = 0
        dataIdx = 512
#3  0x00007fffefa33c55 in BinaryRecordingEngine::BinaryRecording::writeData (this=0x5b15500, writeChannel=0, 
    realChannel=0, buffer=0x7fffe860615c, size=1024) at BinaryRecording.cpp:159
        multFactor = 0.00015650517889298499
#4  0x00000000005b8c3b in RecordThread::writeData (this=this@entry=0x11fe540, dataBuffer=..., 
    maxSamples=maxSamples@entry=4096, maxEvents=maxEvents@entry=32, maxSpikes=maxSpikes@entry=32, 
    lastBlock=lastBlock@entry=false) at ../../Source/Processors/RecordNode/RecordThread.cpp:124
        eng = 0
        chan = 0
        timestamps = {data = {<juce::DummyCriticalSection> = {<No data fields>}, elements = {data = 0x7fffb8043090}, 
            numAllocated = 32}, numUsed = 32}
        idx = {data = {<juce::DummyCriticalSection> = {<No data fields>}, elements = {data = 0x7fffb8041c50}, 
            numAllocated = 32}, numUsed = 32}
        events = std::vector of length 0, capacity 0
        spikes = std::vector of length 0, capacity 0
#5  0x00000000005baa20 in RecordThread::run (this=0x11fe540) at ../../Source/Processors/RecordNode/RecordThread.cpp:97
        dataBuffer = <optimized out>
        closeEarly = false
#6  0x0000000000773d77 in juce::Thread::threadEntryPoint (this=0x11fe540)
    at ../../JuceLibraryCode/modules/juce_core/threads/juce_Thread.cpp:102
        currentThreadHolder = {referencedObject = 0x7fffd80008c0}
#7  0x0000000000773f69 in juce::juce_threadEntryPoint (userData=<optimized out>)
    at ../../JuceLibraryCode/modules/juce_core/threads/juce_Thread.cpp:117
No locals.
#8  juce::threadEntryProc (userData=<optimized out>)
    at ../../JuceLibraryCode/modules/juce_core/native/juce_posix_SharedCode.h:903
No locals.
#9  0x00007ffff6c536ba in start_thread (arg=0x7fffdfffe700) at pthread_create.c:333
        __res = <optimized out>
        pd = 0x7fffdfffe700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140736951478016, 4134503634717544639, 0, 140737488343967, 
                140736951478720, 0, -4134574002361954113, -4134488436723904321}, mask_was_saved = 0}}, priv = {pad = {
              0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#10 0x00007ffff5ee082d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
No locals.
(gdb) 
```
As you can see, `startPos=18446744071562067456`, a rather big number. Note that `18446744071562067456 / (1024**6) = 16.0`. Not sure what this signifies...

aacuevas, 2017-03-16:
That value in startPos is normal, since it specifies the position in the file to start writing at, which for 128GiB needs to be rather large.

I have a couple of clues of where the issue might be but I need to think a good way to debug it without needing to write tons of lines to the console for hours, thanks for the detailed gdb output!

This is silly to ask, but to discard obvious things, your system is running a 64bit kernel, right?

Aarón.

mspacek, 2017-03-16:
Yes, it's running 64-bit Xubuntu 16.04.

mspacek, 2018-05-16:
This is an old issue, and I haven't managed to get the latest dev running with our headstage (see #219), but I re-ran the above dev version of open-ephys (from March 2017) with 67 channels instead of 32, and again at just under 20h of recording, I get a segfault. The output .dat file ended up being exactly 268.0 GiB. Some math:

original issue with 32 chans:
`ntimepoints = 128*1024**3/2/32 = 2147483648`
new result with 67 chans:
`ntimepoints = 268*1024**3/2/67 = 2147483648`

In both these cases, the number of timepoints written to disk is exactly `2**31`. So I don't think it's data volume that's the issue, but rather timepoint count. There's probably an int32 somewhere that's overflowing, perhaps within JUCE?

mspacek, 2018-05-24:
I just tried it on the latest `master` branch, and the bug exists there as well. It segfaults at `2**31` timepoints. This time I had 64 channels of acquisition, which led to a .dat file of exactly 256.0 GiB:

`ntimepoints = 256*1024**3/2/64 = 2147483648`

mspacek, 2018-06-09:
I just realized that the timestamps provided by the Rhythm node are 32 bit, so big surprise things stop working after ~20 hours at 30 kHz. From `rhd2000evalboard.cpp`:

```C++
// Returns the number of 16-bit words in a USB data block with numDataStreams data streams enabled.
unsigned int Rhd2000DataBlock::calculateDataBlockSizeInWords(int numDataStreams, bool usb3, int nSamples)
{
	unsigned int samps = nSamples <= 0 ? SAMPLES_PER_DATA_BLOCK(usb3) : nSamples;
	return samps * (4 + 2 + numDataStreams * 36 + 8 + 2);
    // 4 = magic number; 2 = time stamp; 36 = (32 amp channels + 3 aux commands + 1 filler word); 8 = ADCs; 2 = TTL in/out
}
```
Is the physical clock itself limited to 32 bits, or is that something that can be modified in the FPGA with a different `.bit` file? I'm not even sure if the physical clock here is the one on the RHD2164 (in our case) headstage, or the acquisition board.

The magic number at the start of each sample is quite long. Perhaps we could cut it down to 32 bits (two 16 bit words), and increase the timestamp field to 64 bits (four 16 bit words)?

mspacek, 2018-06-09:
Actually, it looks like the timestamp in the USB data frame is returned as a unint32, so the timestamps should last for ~40 hours instead of 20 hours at 30 kHz. Maybe the underlying data in the usbBuffer is actually a signed int32 and maxes out at 2**31?

```C++
unsigned int Rhd2000DataBlock::convertUsbTimeStamp(unsigned char usbBuffer[], int index)
{
    unsigned int x1, x2, x3, x4;
    x1 = usbBuffer[index];
    x2 = usbBuffer[index + 1];
    x3 = usbBuffer[index + 2];
    x4 = usbBuffer[index + 3];

    return (x4 << 24) + (x3 << 16) + (x2 << 8) + (x1 << 0);
}
```

aacuevas, 2018-06-09:
This is embarrassing. 
I've been so sure all this time that Rhythm timestamps were 64bit that I've ignored all documentation and pieces of code actually saying otherwise.
Changing the data frame is not trivial. Replacing 32bit from the magic word to the timestamp would be an option, but the reason it is that long is to reduce the chance of that pattern appearing randomly if the blocks were unsynched, leading to data corruption. I'll study the issue and weight our options.

Not sure why it causes an issues at 2**31, though. The data inside the FPGA is definitely unsigned and thoughout the buffering methods it is always referred to as unsigned.

jsiegle, 2022-07-30:
Sample numbers within the GUI are now all stored as signed 64-bit integers, and block addresses in sequential block files are all unsigned 64-bit integers. So this problem should be fixed.

closed 2022-07-30 by jsiegle