open-ephys/plugin-GUI #124: Some questions about the auxilliary input sampling rate and sampling order

galenlynch, 2017-03-16:
I'm attempting to use the auxiliary input voltages recorded by the open ephys plugin gui, and am having a hard time reconciling my understanding of the RHD2000's operation with the .continuous files produced by the Open Ephys  GUI. 

My understanding is that a single auxiliary input to the RHD2000 chips is sampled on three of every four SPI command sequences, resulting in a sampling frequency for each AUX input that is 1/4 of the neural sampling frequency. However, the .continuous files for the aux channels report the same sampling frequency as the neural channels. When I look at the sample values for the auxiliary files, it looks like each ADC value is simply repeated four times to hide this difference in sampling rate. Is this correct?

I am also interested in the order of the samples from each auxiliary input, as I am sampling the same signal source with all three auxiliary inputs. The .continous files for auxiliary channels have coordinated jumps of the measured voltages for all auxiliary inputs to the same chip. My understanding is that the RHD2000 is actually sampling the auxiliary inputs in sequence. After looking at the plugin gui and rhythm source code, it looks like AUX1 is sampled during the first element of each four sample block, AUX2 is sampled during the second element of each four sample block, AUX3 is sampled during the third element of each four sample block, and no input is sampled during the fourth sample of each four sample block. Is this correct?

Thanks for your time!

aacuevas, 2017-03-16:
Hi,

Yes, both statements are correct. Since at this moment the GUI source nodes can only output data at a single sample rate it was needed to replicate the aux channels that way. Newer versions of development have the ability to send data at different rate so we might change that in the future, but I'm not sure yet.

galenlynch, 2017-03-16:
Great, thanks for the clarificaiton!

closed 2017-03-16 by galenlynch