open-ephys/plugin-GUI #125: Pluging gui produces corrupt .continuous files for plugin gui 0.4.1

galenlynch, 2017-03-18:
I have noticed that for some of my recordings, the .continuous files from the same recording have different sizes on disk, corresponding to different numbers of complete data blocks. Often, the smaller files cannot possibly be comprised of complete data blocks. Even before the truncted block in a damaged file, the surviving blocks have defects such as an incorrect end or record marker (all zeros instead of the sequence specified), or other problems.

aacuevas, 2017-03-22:
Hi,

This is the fitst notice we have of similar issues with the open-ephys format. Even when blocks are missing the basic file structure should be stable. 

Do you know if the file corruption happens randomly or is there any correlation with another event?
What are the OS and the computer specs you're using?
Could you please send me ( aaron@open-ephys.org ) the settings.xml file and one of the corrupted .continuous files?

Aarón.

galenlynch, 2017-04-26:
Sorry that it took me so long to respond to this. I sent you the corrupt .continuous file and settings.xml file. I can reliably produce corrupt .continuous files by taking short recordings.

jsiegle, 2017-04-26:
How short is "short"? I think we've noticed a few problems related to recordings <2 minutes, presumably due to the buffering that happens inside the recording thread.

aacuevas, 2017-04-26:
I was about to ask the same question. In theory the recording system should work with short recordings, I've tested it with a couple of seconds, but there might still be some corner cases which cause this corruption. I'll take a look at the files and see if I can figure out something from them.

galenlynch, 2017-04-26:
The file that I just sent @aacuevas was 22 seconds long. I routinely take short recordings (10-20s) when testing for antidromic activation of neurons, which is mostly when I see this problem.

galenlynch, 2017-04-27:
I was looking at one of the corrupt files and realized that the timestamps in the corrupt files start 256 samples after the timestamps in the good files.

jvoigts, 2017-08-24:
This is likely a duplicate of https://github.com/open-ephys/plugin-GUI/issues/154 - could you check if the recent bugfix resolved this and close if that's the case?

galenlynch, 2017-08-24:
I will check in the next week. I also routinely turn off channels, it could very well be the same issue as #154.

closed 2020-12-15 by medengineer