open-ephys/plugin-GUI #126: data timestamp misalignment

mikewehr, 2017-03-24:
We are having a problem where the data timestamps are not continuously increasing. Rather than steadily increasing in steps of 1/samplerate, there are regular discontinuities such that timestamps can increase or decrease. This occurs consistently every 1024 samples. The increases or decreases are one of a small set of possible values (+15.9 ms, +25.9 ms, -4.1 ms, -14.1 ms, others) and appear to vary somewhat randomly between these values every block. It happens for every single data acquisition, but with a different set of time shift values for any particular file. Because there are both increases and decreases, the net time error accumulation is small, but they do not exactly cancel out. 
We experience this problem with latest plugin GUI compiled on Ubuntu, but it doesn't happen with pre-plugin versions, nor with with the latest plugin GUI compiled on Windows. We initially suspected underbuffering due to resource load, but it happens even when GUI is the only application running, and with the lightest possible configuration (only one continuous channel).
plugin GUI 0.4.1
Ubuntu 14.04, intel i7-6700k 4GHzx8, 32 GB ram

Has anyone else seen this? Any idea what's going on?
It looks very similar to the issue posted by kkke on Sep 15, 2016, but I didn't see any solution there. [https://github.com/open-ephys/analysis-tools/issues/32](url)
many thanks,
Mike

Here is an examples of a data file that illustrates the problem:
https://drive.google.com/file/d/0B7Iw_28oHZ9gRkhWSTlwS0JjV2s/view?usp=drive_web

Here is an example of matlab code that illustrates the problem for this dataset:
[data, timestamps, info] =load_open_ephys_data('105_ADC1.continuous');
figure
hold on
plot(1:length(timestamps)-1, diff(timestamps), 'o-')
xlim([0 100000])
plot(0:1024:100000, -4.1e-3, 'r*')
plot(0:1024:100000, -14.1e-3, 'r*')
plot(0:1024:100000, 15.9e-3, 'r*')
plot(0:1024:100000, 25.9e-3, 'r*')

mikewehr, 2017-03-25:
By the way, load_open_ephys_data_faster produces the same results.

jonnew, 2017-03-25:
I'm fairly sure I've experiences this as well. I started a write up about it and go distracted.

gaelias, 2017-06-15:
Not sure if anything's been done on this issue but I've come across it as well. From what I've seen looking into it it seems to only affect nodes that are present after the splitter. Below is an example file I recorded where I'm plotting the trace from the channel map in green, red and blue are traces from bandpass filters after a splitter, blue = unit filter (300-6k), red = LFP filter (1-300; chan map & LFP filter are rescaled to fit in the same window). The black boxes highlight areas where the timestamps are discontinuous for the LFP and unit traces (nodes after the splitter) but not for the channel map (node immediately prior to the splitter).  
[image]
When I plot the LFP timestamps vs the channel map timestamps the difference is pretty obvious. Here's an example of that plot, zoomed in so you can see the difference. The channel map timestamps increase linearly while discontinuities arise in the LFP timestamps causing them to drift away from the diagonal (note I did not zoom in perfectly to get this shot so the lines are not perfectly situated across the diagonal of the plot window.
[image]

This effect is also obvious when examining the data around one of these episodes and comparing plots created where their associated timestamps or their index value are the x-axis values. In these plots the top trace (blue) is the raw trace (I made these figures at different times so please forgive the change in color scheme, also, this trace is pulled directly from the FPGA, not channel map but the channel map gives a nearly identical trace), the middle (red) is the LFP trace where I've sorted the data based on their associated timestamps (I thought maybe the timestamps were just being read out of order), and the bottom (black) trace is the unsorted LFP trace. Note how in the "x-axis = timestamps" figure the trace appears to loop back on itself in the bottom plot. If you were to zoom in on the middle plot (time sorted) you would see the trace jackknifing up and down during this period. When the data is replotted based on index ("x-axis = index" figure) it shows how selection using timestamp ranges pulls a larger number of data points in the filtered trace compared to the raw. Another thing to note in this plot is how silimar the unsorted trace looks compared to the raw trace at comparable time periods. To me this suggests that the actual data is being recorded fine it's just the timestamps that are messed up either in their loading or recording.
[image]

[image]

Finally, I've also played around with recording just LFP data (sans splitter) and I did not see any temporal discontinuities in those traces. However, for those recordings I was also recording at a lower sample rate than I am here (5k for LFP alone vs 30k for the data shown above). Hopefully this all helps provide some clue as to what's going on.

mikewehr, 2017-06-15:
We decided to workaround this problem by switching to windows, which worked for us. Gabriel, are you using linux?

gaelias, 2017-06-15:
Nope, I do everything with windows machines. Did switching to windows rectify this for you?

jsiegle, 2017-06-20:
Can people provide some more details about the hardware and signal chain that produced this issue? I'm trying to replicate it on Linux (CentOS) with the latest `development` branch plugin-GUI, using a 32-channel headstage. So far the timestamps look identical for processors before and after a splitter.

mikewehr, 2017-06-21:
plugin GUI 0.4.1
Ubuntu 14.04, intel i7-6700k 4GHzx8, 32 GB ram
32-channel headstage.
we observed it with all permutations of signal chains that we tried, including the most basic possible.
Switching to windows completely solved it for us.

gaelias, 2017-06-21:
plugin GUI 0.4.1
Windows 10 Home 64-bit (10.0, Build 14393), i7-6700HQ 2.6GHzx8, 12GB ram

Chain:
FPGA (30k)->ChanMap->Splitter->[BandpassLFP->LFPview] / [BandpassUnit->SpikeDetect(tetrodes)->SpikeView]

For the first figure I posted the data was collected recording 64 channels, the rest are from 96 channels.

I only saw the issue when I was recording using the splitter. When I removed the unit filter/recordings and only recorded LFP I didn't have this issue although I simultaneously dropped my sample rate (to 5k) when I stopped recording unit data for my most recent recordings.

jvoigts, 2017-08-24:
This is likely a duplicate of https://github.com/open-ephys/plugin-GUI/issues/154 - could you check if the recent bugfix resolved this and close if that's the case?

ethanbb, 2017-10-24:
Just want to confirm that I was experiencing the same issue, specifically when recording non-contiguous blocks of channels (e.g. 1-8 and 17-24, but not 9-16), and upgrading to a version after the fix to OriginalRecording resolved the issue.

closed 2020-12-15 by medengineer