open-ephys/plugin-GUI #127: RecordControl module "freezes" acquisition

malfatti, 2017-04-14:
Hello!

I'm running experiments that require several short recordings (~100 recs of 2 seconds each) on a data file. I set the RecordControl to toggle recording on and off at the rising edge of an input, and send square pulses to a digital input channel. 

In the past I used this module to toggle one to three recordings per file, so I never any issue. Now that I have to toggle more than 100 recordings per file, I noticed that ~25% of the trials the RecordControl toggles the recording on and "freezes" the acquisition. The interface still works, time continues counting, there are no errors in the terminal, but it does not detect any further events, so it never stops the recording and of course never starts a new one. If I stop the recording manually,  it stops and closes the file correctly, and detect the next event correctly. In rare occasions I stopped the recording manually and the entire GUI froze, and the data file was not closed properly. This happens even more often if I add an LFPViewer.

In my system, this affects master, testing and development. Interestingly, I don't have this problem with the old non-plugin GUI, even using complex signal chains. I think this shows that this issue is not related with missing libraries, audio setup or OS. Also, I have this issue with both OpenEphys and Kwik files.

Steps to reproduce:
- Add FPGA and RecordControl modules;
- Set RecordControl to toggle at the rising edge;
- Run acquisition and toggle at least 20 recordings using the RecordControl;
- Try also with an LFPViewer, its window freezes completely and restore when rec is manually turned off.

Is this affecting only me?

Cheers :D

jsiegle, 2017-04-14:
Can you adjust your experiment design to record continuously instead? You can use the toggle channel as a TTL input to mark where the beginning and end of each recording would have been. My guess is that this solution will be a lot simpler than trying to debug this issue. It's likely related to the fact that recording happens in its own thread, which helps distribute CPU load under normal circumstances, but won't deal well with very short recordings.

malfatti, 2017-04-15:
Sure :) I'm already using this workaround. Thanks for the feedback.

Cheers :D

closed 2017-04-15 by malfatti