open-ephys/plugin-GUI #131: unable to load settings with selected recording channels OEv2 acquisition board

lepmik, 2017-04-28:
When saving a signal chain with only a few selected recording channels the gui crashes after loading it and starting acquisition. Unable to get any clue to the reason as the gui freezes completely.

Windows 7 enterprise
Visual studio 13 release 64 build
usb3 cable in ss slot to acquisition sys

lepmik, 2017-05-07:
In addition it is crashing when loading a settings file with deselected TTL channels in the LFP viewer.

ethanbb, 2019-01-17:
I can't replicate this (using USB2). Is it still happening for you with the latest release?

jsiegle, 2020-12-22:
This is no longer relevant, now that the recording settings all live in the Record Node (rather than individual plugins).

closed 2020-12-22 by jsiegle