open-ephys/plugin-GUI #133: Recording index

shayo, 2017-05-22:
It's very useful to know the current recording index when running experiments (i.e., writing in your notebook that a particular experiment happened in a specific recording).
I think at some point this was visible on the right region (next to the timestamp)? But it is no longer there.
Aaron, can you please add some visible indicator of the current recording record number, or check why it is not showing? (at least not on the plugin dev branch).
Thanks!

Shay

aacuevas, 2017-05-22:
I don't recall there ever being an indicator of the current recording number. There isn't any related code at all in the branch at least.

In any case, you're right in that it is quite useful to have it visible somewhere in the GUI. I'll add a display for it.

jsiegle, 2022-07-30:
You can see the current recording index by looking at the data directory, or query it remotely via the built-in HTTP server. However, it's not actually visible anywhere in the GUI itself, so I'm going to leave this one open.