open-ephys/plugin-GUI #134: Errors Debugging/Building GUI in Visual Studio 2017

edbhan, 2017-06-02:
Initially, upon following the open-ephys Wiki instructions to download and build the GUI and associated plugins, I encountered the error message prompting me to install v141 build tools. I went in to the .vcxproj file and changed all of the incorrect build tools to 141. This seemed to work, and the code began to run in VS2017 without the v141 build tools error.

However, now there are numerous (>3000) error messages that appear when the debug proceeds, the majority of which are either syntax errors or "function returns function" errors. Obviously, the GUI fails to build properly because of these errors. I'm curious whether or not I can build the GUI with VS2017, or if there are other steps I should take to build the GUI properly. 

Operating System: Windows 10
Visual Studio 2017 (w/ 2015 also already downloaded)

AhmadKhatoun, 2017-09-20:
Hi edbhan. AM facing the exact same problem. Did u figure out how to solve this issue?

edbhan, 2017-10-02:
There was an undergrad working on that this summer that sent out the
email.  I think he tried several simple things that did not work.  Sorry I
don't know the exact details.  When we get the chance, we'll investigate
more fully and report back.  Good luck!

nicojosset, 2018-08-16:
I am having the same issue!

Did anyone find a solution?

angelonc, 2018-09-05:
For what it's worth: The guide recommends using VS 2013 to build it, which is what I've used, and it has worked fine. Specifically, I downloaded Visual Studio 2013 Community.

Unless there's a pressing reason for using 2017, using 2013 might be your best bet.

medengineer, 2020-12-15:
Fixed w/ new CMake architecture

closed 2020-12-15 by medengineer

Barryjuait, 2021-05-27:
Hi @medengineer , I tried the 3.20.2 release of cmake, but it didn't work. What am I missing?