open-ephys/plugin-GUI #135: Main branch under mac - no files are saved to disk

shayo, 2017-06-14:
Hi,
I have a weird issue on my mac. When I click record, only the xml file is generated. Nothing else is saved to disk. This is happening on the master plug in branch.
It is reproducible. Seems like the first time I ran the exe, it worked, but then subsequent calls suffer from this problem.

shayo, 2017-06-14:
It's even worse. When I click record, the stream stops and nothing is updated on the screen...

aacuevas, 2017-06-14:
Hi,

Does it happen with both KWIK and open ephys formats or just the latter? How many channels are you trying to save?
Also, could you please check if this also happens on the testing and/or development branches? I do remember some patches regarding recording since last master update, but I'm not sure if they might be relevant to your issue.

Aarón.

jsiegle, 2020-12-16:
I've been testing extensively on Mac, and this doesn't seem to be a problem anymore.

closed 2020-12-16 by jsiegle