open-ephys/plugin-GUI #139: Adding Network Events causes system freeze upon start record in Plugin-GUI master

shayo, 2017-06-26:
I have a very simple chain (see attachment), but it seems like when I add network events and then hit record, the system freezes on OSX.
Removing the Network Events PlugIn solves the freezing, so I suspect it happens there.

Can someone check / reproduce?

[not_working.txt](https://github.com/open-ephys/plugin-GUI/files/1102500/not_working.txt)

shayo, 2017-06-26:
This doesn't happen on the GUI-master branch.

zekearneodo, 2017-08-13:
Did you figure a workaround for this? I am having this issue (on OSX), using the master branch of plugin-GUI.

shayo, 2017-08-13:
No. I switched to a Windows machine :(

c-wilson, 2017-09-13:
Hey, I think this is a duplicate of #67 and is still an issue, but there is a workaround in that thread.

zekearneodo, 2017-09-30:
Indeed, that workaround works.
Thank you.

jsiegle, 2020-12-16:
Just tested on Mac, and this is no longer happening.

closed 2020-12-16 by jsiegle