open-ephys/plugin-GUI #140: NetworkEvents is broken on main branch

shayo, 2017-06-27:
timestamps are not added to the midi buffer properly. This can be traced back to GUI branch, in which timestamps were removed from the message that is placed in the buffer.
I see that some changes have been made to the development branch about this, but it would be nice to get this fixed on the stable branch until the development one is ready.

aacuevas, 2017-06-28:
What do you mean with the timestamps not being added properly? If I recall correctly, the network events originally saved within the string a software timestamp (measured in system ticks), which was completely independent of the timestamp of the event subsytem (which in most cases for the network event means that all events received within a processing block are timestamped with the start timestamp of said block). 
At some point we removed the software timestamp because it was being saved as a unreadable binary string into the messages text file and some users regarded it as a bug, since the hardware timestamps were enough for them. Do you mean that? It is easy to just reenable copying the timestamp into the midi buffer, but in any case I'd convert it to string first.

With the new event subsytem in the development branch that will not be necessary. The system timestamp is interpolated taking into account both the software and hardware times, so the event can be timestamped in terms of hardware times, so events are easier to synchronize with everything else.

shayo, 2017-06-28:
The timestamp was used with the PSTH module to know when network messages
arrived to align spikes properly.
It took me a while to understand why events were not registered properly...

Yes, I saw the recent changes on the development branch. However, I can't
use that one until recording issues with OSX are solved.

Shay Ohayon
HHMI Postdoc Fellow of the Life Sciences Research Foundation
DiCarlo's lab, MIT, 46-6147
+1 617-324-3593

jsiegle, 2022-07-30:
The source code for the Network Events plugin now lives here: https://github.com/open-ephys-plugins/network-events

If you're experiencing any problems with this plugin, please open an issue in that repository.

closed 2022-07-30 by jsiegle