open-ephys/plugin-GUI #141: plugin-gui development : plugins.sln doesn't compile under visual studio 2013

shayo, 2017-06-27:
multiple errors complaining about a missing files "open-ephys.lib" and "OpenEphysHDF5Lib.lib"

aacuevas, 2017-06-28:
The error with x64 is due to some path not being properly set. They should be (I develop with VS2013 and mostly build as x64) but I'll check the config files.

As for the second, try opening the project properties and going into "Common Properties->C/C++->General" and force disable the SDL checks option. It should be disabled by default but VS sometimes does weird things. The error is due to some methods being marked as deprecated by JUCE when we updated to JUCE4, we need to check if we can safely remove those without other things breaking apart.

shayo, 2017-06-28:
The two projects that fail to compile properly are the NWB and KWIK plugins.
After manually adding the debug folder (which contains OpenEphysHDF5Lib.lib) I am getting many errors about unresolved external symbols associated with HDF5.

shayo, 2017-06-28:
master branch also fails to compile with multiple errors about missing open-ephys.lib

shayo, 2017-07-31:
What's the latest on this? I tried to d/l the latest master branch again yesterday and it still will not compile under VS 2013.
Please try to do this on a fresh windows installation to see the issues I'm talking about.

I made a couple of changes I'd like to push back at some point and it would be easier to do a PR on the latest version that actually compiles.

Thank you!

Shay

shayo, 2017-07-31:
i think I realize where the confusion originates (mix-up between Release and Release64)
You might want to make these names compatible
(i.e., compiling the GUI needs configuration Release64 for x64, while compiling plugin for x64 is just called "Release")

aacuevas, 2017-07-31:
I've just rechecked with a clean repo and while the development branch compiles with no issues, the master one had a weird configuration error that made the plugins to always be compiled in release mode, even when debug was selected, which could cause some of those library issues.

I've fixed that and now both master and development build just fine for me on VS2013, please check if that is the case for you too.

shayo, 2017-07-31:
See my previous comment. You might want to change "Release" to "Release64" to avoid confusion between compiling for x32 and x64 (this is why it didn't find the open-ephys.lib file in my case).

closed 2017-07-31 by shayo

aacuevas, 2017-07-31:
Oh, I had github open and didn't realize there was another message.
The thing is, in VS you have on dropdown to select between Release and Debug, and another to select x32 or x64 architecture, which is the proper way for visual studio to manage configurations.
However, for some reason I yet don't understand, Projucer is unable to generate proper configurations for multiple architectures, which forced us to configure it so it creates two different release and debug configurations for the core GUI, which in turn also require to select the proper architecture (If you select Debug64 you also need to manually select x64 still, you could select Debug64 with x32 architecrure selected and it wouldn't properly compile).
While this is done automatically by projuce for the core, the plugin projects are a different history and having to manage four different configurations and select which compile in which architectures would be a maintenance nightmare.
What we can do is check the documentation and be sure that this is clear enough.