open-ephys/plugin-GUI #142: plugin dev branch: Spike raster plugin crash

shayo, 2017-06-28:
As soon as I hit the maximize window (to open the plugin UI), the entire system crash.

jsiegle, 2017-06-28:
This plugin is still under development, and it hasn't been updated to work with the new channel objects yet. Expect lots of crashes until it's fully tested.

jsiegle, 2020-12-16:
The Spike Raster plugin was never finished, but there are now two options for viewing spikes across many channels:

- The LFP Viewer can set a "Spike raster thresh" value, which will display any time points that cross this threshold as vertical lines
- The Probe Viewer plugin can operate in "Spike rate" mode, which shows the spike rate across channels plotted as a heat map

closed 2020-12-16 by jsiegle