open-ephys/plugin-GUI #144: Extracting sample rate from settings.xml

malfatti, 2017-07-08:
Hello!

I'm trying to get the sample rate used in recording from the settings.xml file. At `/SETTINGS/SIGNALCHAIN/PROCESSOR/EDITOR/SampleRate` I'm getting "17" for 30kHz recs. Is there a way to extract the sample rate used in recording from the settings.xml file?

Cheers :)

angelonc, 2018-02-22:
Bumping this because I'm having the same issue as well!

jsiegle, 2018-02-23:
I just pushed an update to the `development` branch that outputs the sample rate as a string to the settings.xml file. "17" refers to the index of the selected sample rate in the dropdown menu, which is required to re-load the previous settings, but is not very helpful for figuring out what the sample rate actually was.

closed 2020-12-15 by jsiegle