open-ephys/plugin-GUI #147: Save Audio Settings with "Save" or "Reload on startup"

ethanbb, 2017-07-24:
Currently, there's no way to save any changes one makes in the Audio Settings dialog box (Audio device type, buffer size, etc.). It would be useful if this were saved along with other signal chain information when you either save the settings to a file or select "Reload on startup."

This might cause some issues, since the settings there seem to be highly system-dependent, so there would have to be some way to check that saved settings are compatible with the current system before loading them. However, I think it is an important feature, since buffer size can be important in some signal chains, and it is easy to forget to change the audio settings each time you start the program.

aacuevas, 2017-07-26:
This is a good point, but as you said it might be complicated to correctly check that the settings are compatible. Perhaps we should add a warning window if they aren't compatible so the users can be aware of that.

godwincharan, 2017-09-03:
Juce provides a way for storing the application settings in user settings. 
We can store these settings. Especially we are storing the Audio settings which are different different system it will never be same. ( Mac, Windows or Linux)

We can use something like this:
ScopedPointer<XmlElement> audioState (deviceManager.createStateXml());
getAppProperties().getUserSettings()->setValue ("audioDeviceState", audioState);
getAppProperties().getUserSettings()->saveIfNeeded();

Please let me know if this works. I will implement and give pull request.

jsiegle, 2017-09-03:
That sounds like it could work. As @aacuevas mentioned, users will be sharing settings files across platforms, so there will have to be some way of checking whether the audio settings are compatible with the current system upon loading.

godwincharan, 2017-09-05:
there are different attributes for the Audio settings like following.
```
<DEVICESETUP deviceType="CoreAudio" audioOutputDeviceName="Logitech USB Headset"
                 audioInputDeviceName="Logitech USB Headset" audioDeviceRate="44100"
                 audioDeviceBufferSize="1024"/>
```
Device type attribute is not going to same for Windows and Mac. In my case its showing the "CoreAudio" in Windows it might be Windows Audio. (It can also be ASIO and other device type)

So porting one machine's audio device settings to other is not ideal. Also other aspect will vary from device to device. Some device will have 8k,16k sample rate but others will not have it. 

Even if we try to select the device with sample rate matching; we can not guarantee that number of inputs and outputs will match.

ethanbb, 2017-09-05:
So if you try to set the device type to something that doesn't exist on the current system, or otherwise provide incompatible settings, do you know where an error would occur (hopefully somewhere)? Is it something that could be handled in a "catch" block that would revert to default settings and show a warning?

P.S. thanks for working on this!

godwincharan, 2017-09-05:
Juce initialises the audio device manager using following API

```
String error = deviceManager.initialise(0,  // numInputChannelsNeeded
                                            2,  // numOutputChannelsNeeded
                                            0,  // *savedState (XmlElement)
                                            true, // selectDefaultDeviceOnFailure
                                            String::empty, // preferred device
                                            0); // preferred device setup options
```
Same API is used in AudioComponent's constructor. If you see the 4th argument to the API is selectDefaultDeviceOnFailure. So If the settings with XML fails it will initialise with default device. 
If that too fails it will return error. (May be we dont have any Audio device at all or all the audio devices are disabled from system preferences);

jsiegle, 2017-09-11:
It sounds like that method should work for reloading settings from the same computer (which is what will be happening most of the time, anyway). If it does succeed, then we can figure out how it fails when loading a config file from another computer. Can you test it out?

ethanbb, 2018-01-16:
I've noticed this seems to be fixed in recent versions - can someone confirm?

jsiegle, 2018-01-16:
It's saving the buffer size, but not any other settings. Is that what you're seeing?

ethanbb, 2018-01-16:
Yup, that would match what I'm seeing. It seems like that might be the best
way to do it, since buffer size might be able to transfer across platforms,
whereas trying to match other settings might break things and (I think)
wouldn't be as important. Does it save buffer size in units of samples or
time?

Thanks,
Ethan

jsiegle, 2018-01-16:
I believe it's in units of samples...so if the config file were loaded on a machine with a different audio card sample rate, the buffer size would change. Fortunately, most default to 44.1 kHz.

ethanbb, 2018-01-16:
OK. With the notable exception of Windows Audio, which is 48 kHz... So
maybe we can keep this issue open since it's still not a perfect solution,
but still thanks for mostly solving this!

closed 2020-12-15 by jsiegle