open-ephys/plugin-GUI #148: libxrandr-dev needs to be added to linux dependencies

cxrodgers, 2017-07-26:
This seems to be required by juce_gui_basics.cpp but is not in install_linux_dependencies.sh.

> Compiling juce_gui_basics.cpp
> In file included from ../../JuceLibraryCode/juce_gui_basics.cpp:9:0:
> ../../JuceLibraryCode/modules/juce_gui_basics/juce_gui_basics.cpp:104:37: fatal error: X11/extensions/Xrandr.h: No such file or directory
>    #include <X11/extensions/Xrandr.h>
>                                      ^
> compilation terminated.

Installing libxrandr-dev using the package manager seems to fix this. Can we add it to install_linux_dependencies.sh? I can create the PR if that would be helpful.

jsiegle, 2017-07-26:
Yeah, a pull request would be great!

cxrodgers, 2017-07-27:
https://github.com/open-ephys/plugin-GUI/pull/149

closed 2017-07-27 by cxrodgers