open-ephys/plugin-GUI #150: Linux installation documentation out of date

cxrodgers, 2017-07-27:
I recently re-installed open-ephys plugin-GUI from scratch and noticed a few things that appear to be out of date.

In general are these instructions supposed to apply specifically to the plugin-GUI, not the old GUI? I assume so since all new users are recommended to install plugin-GUI.

- There are multiple references to an "ubuntu_14.04" branch which I think no longer exists (or is in the old non-plugin GUI). Can we remove these references?
- There is a Builds/Linux/setup.sh script which performs most of the installation but is not referenced in the documentation.
- Are the ZeroMQ and HDF5 installation instructions still necessary with the plugin version of the GUI?
- setup.sh uses "-j4" and the documentation does not. Probably doesn't really matter but I think we should be consistent. Which is recommended for inexperienced users?

Finally the make process produces a binary called "open-ephys.so", which I think is an error. It should be called "open-ephys", right? The binary runs, but I think this is not conventional naming, and also it looks like "setup.sh" will fail because it assumes it is called "open-ephys".

aacuevas, 2017-07-31:
Hi,

It seems indeed that the documentation needs a urgent revision. Answering your points:

- Yes, it is safe to remove the references to the old branch.
- It is not? That is a huge oversight on our part, since that script takes care of installing the required dependencies.
- ZMQ and HDF5 are still used by some plugins that are built by default, so unless users manually disable those plugins, the libraries are still needed.
- What the -j4 option does is use four different threads to compile the software faster. It has none other effects but build speed, so it does not really matter if it is set or not. Traditionally, linux build instructions never take that flag into account, since it is left to the user to decide how many threads or other make options to use, thus just describing the base command, while setup scripts do use some default options. We just adhered to that convention, but it should not harm to display the -j4 option in the documentation, since most computers do have multiple cores nowadays anyway.

The .so is a known issue, which I suspect stems from a bug in the Projucer software. I tend to manually modify the makefile to change that whenever I notice, but if the project is rebuilt by Projucer it would revert back to that state, requiring a manual change of the makefile again. What we could do while seeking  more permanent solution would be adding some checks to the setup.sh script to automatically rename the file if needed.

cxrodgers, 2017-07-31:
Thanks for the info and explanation! I'll make some changes to the wiki accordingly.

One clarification: there are two Linux installation scripts that do different things:

- Builds/Linux/setup.sh -- this calls "make" and links to /usr/bin
- Resources/Scripts/install_linux_dependencies.sh -- this installs dependencies

The first one is the one that is not referenced in the documentation, but in any case, I think it's probably preferred to describe the make commands explicitly and let the users type them in. I personally don't choose to install to /usr/bin anyway.

aacuevas, 2017-07-31:
Oh, you're right, I mixed up the scripts in my head.
Having the script might still come in handy, but I'm also fan of letting users type the commands themselves.

cxrodgers, 2017-08-03:
Thanks for this information, I've updated the wiki documentation accordingly.

closed 2017-08-03 by cxrodgers